betaflight/betaflight #2127: OMNIBUSF4SD LED Strip not working

TobiFPV, 2017-01-15:
On Omnibus F4 AIO V2 (with SD and Baro) the LED-Strip is not working when i use the Target OMNIBUSF4SD. I tried on PWM5 and the dedicated LED Pin (J9).
If i use target OMNIBUSF4 the LED-Strip works fine on PWM5.

jflyper, 2017-01-15:
@TobiFPV 
For OMNIBUSF4SD targets, LED strip is on PWM6.
It has been moved because the PWM5 had DMA conflict with SPI used for SDCARD.

TobiFPV, 2017-01-15:
Sounds legit. Sorry that i didn't found that out on my own.
Thanks a lot!

closed 2017-01-15 by TobiFPV

dentex, 2022-06-19:
Hi,I have a Hobbywing F4 G2, and in order to use the *ledstrip* I had to type in the CLI
`resource led_strip a8`. I don't know if this collides with the SDcard usage, I don't use it.

dentex, 2022-07-14:
Hello again, I'm still searching for a solution. The buzzer it's one of those with the small LiPO on-board, so I only use the Buzzer- pin. Can this be the cause of the continuous beep if I do `resource led_strip a8`?