betaflight/betaflight #2128: Air Mode does not seem to be working

Gaz-B, 2017-01-15:
This is on Kiwi F4 The OSD and configurator says its functional The blackbox log tells me when its engaged but cant hair or feel any difference either flying or hand held. Motor outputs at idle look the same in BB when moving quad about hand held .This is on build #937 running D-shot 600.

Gaz-B, 2017-01-15:
I seem to have found it if you could confirm in CLI this is the deffaul (set airmode_activate_throttle = 1350) I have change this to (set airmode_activate_throttle = 1000) and can feel it is now functional.

Eganwp, 2017-01-15:
Unfortunately that's the wrong usage of airmode_activate_throttle. Please see wiki. It is only for first time initialization and has no function after throttle passes this threshold.

Gaz-B, 2017-01-15:
Ok i see it's safe guard against activating before flight then I will set it a bit higher.

closed 2017-01-15 by Gaz-B