betaflight/betaflight #2129: CC3D Revolution no longer connects to configurator through UART1

Jahnkeanater, 2017-01-15:
The USB port on my Revo has been broken for the past 6 months so the only way I can connect to it is with a ftdi adapter through UART1.

BF 3.1 RC6 is the latest build that I can still connect to.
Is MSP still on by default on UART1?

AndersHoglund, 2017-01-15:
Nope. Was disabled here: https://github.com/betaflight/betaflight/commit/ccc7292b6c9800eb96fa86745128e11a53df0034

mikeller, 2017-01-15:
No it has been disabled, since people reported problems with the FC going into a reboot loop when they connected a serial RX to UART1 before disabling MSP.

If you need this feature for your hardware, you'll probably have to do custom builds from now on.

Jahnkeanater, 2017-01-15:
I think having MSP on by default on at least one of the UARTS is a good idea. How hard would it be to fix the MSP issue?

If I update to RC8 is it possible to preserve the settings from RC6?

How hard is it to create a custom build? Is there a video somewhere where I can figure out what software I need to install? And in what folder is the code for each targets default settings?

mikeller, 2017-01-15:
@Jahnkeanater: The issue is systemic, and as such not really fixable. If you connect a device that sends data at >= 100kbps without waiting for any feedback (a serial RX) to a port that expects MSP, you will get unexpected results, since some of the very short command sequences will be hit by the 'random' data coming from the RX. 

Building your own binaries is not all that difficult, all you need locally is git and docker, see https://hub.docker.com/r/betaflight/betaflight-build/

Jahnkeanater, 2017-01-16:
I understand the issue but couldn't the configurator automatically disable MSP when serial RX is enabled on that port.

There is nothing stopping someone from enabling both MSP and serial RX on the same port.

blckmn, 2017-01-16:
The issue is the corruption created when someone flashes AND an RX is plugged in on first power up. Normal use case is that VCP is available and works - your case is an edge case. 

In the future we may be able to make the communication for MSP (or rather its replacement) more robust to ensure this issue cannot occur.

blckmn, 2017-01-16:
Can I offer you a FC upgrade? :)

mikeller, 2017-01-16:
> The issue is the corruption created when someone flashes AND an RX is plugged in on first power up.

Exactly. And this is compounded by this configuration usually leading to a reboot loop, thusly making it impossible to change the configuration to something that works without first physically disconnecting the RX from the FC.

> There is nothing stopping someone from enabling both MSP and serial RX on the same port.

If both of these are enabled, and the configuration is stored, an invalid configuration is detected and the settings for this port is reset.

Jahnkeanater, 2017-01-16:
Ok I get the issue now. That makes sense. Would there still be an issue if MSP was on uart3 or uart6 instead? 

The broken FC is a Revo. I wish the thing would die so I had a reason to replace it. :)

mikeller, 2017-01-16:
The problem is the same really. The FC will go into a reboot loop if anything that sends a data 'random' stream is connected to a port that is set to MSP.

I hear hitting the FC with a hammer is a good way to get it to die. ;-P

AndersHoglund, 2017-01-17:
I feel your pain with a half crippled but still usable Revo. I had a bit of the same issue when debugging and flashing CC3D. When flashing over UART1/Main-port it was very convenient to also have MSP on UART1 after reboot. Now with that gone I had to unplug Main and Plugin USB for each test-iteration.

Could we make it a bit easier to build a custom binary with MSP on UART1 by re-inserting the now removed code and enclose it with some conditional compile? Like: `#ifdef USE_MSP_UART1_DEFAULT`   or so. I.e. revert ccc7292 and add a condition around it it.

blckmn, 2017-01-17:
See: 

https://github.com/betaflight/betaflight/pull/2144

Allows for: `make NAZE OPTIONS=USE_MSP_UART`

martinbudden, 2017-02-21:
@blckmn , can this issue be closed now?

closed 2017-02-21 by blckmn