betaflight/betaflight #2131: Multiple IR transponder protocol: How to integrate?

jflyper, 2017-01-16:
There is an implementation for [EasyRaceLapTimer](http://www.easyracelaptimer.com) coming up, and I felt a need for a discussion on how to integrate this with existing iLap facility.

What I can think of is following two ways.

1. Create and use a config/cli variable, say, `transponder_protocol`.  
Use it to switch among different protocols.  
A new MSP should be assigned to manipulate this variable, and configurator should be modified to handle the new MSP message.

2. Prefix the ID word with a single character indicating protocol type.  
For example, "`*35`" for the ERLT, "`#DEADBEEF`" for yet another protocol. Exisiting iLap can continue to use "non-prefixed" ID.
No new MSP message, no configurator change.  
Hardware initialization, at least prescaler and timebase loading, must be postponed until first call to `transponderInit`.

Any thoughts?

rafl, 2017-01-16:
I think option #1 might make for a better user interface, both in the configurator and in the CLI. If we default the new `transponder_protocol` variable to the only previously existing protocol, there won't even be any incompatible changes - you'll just have to upgrade the configurator to change the protocol to one of the newly supported ones.

I suppose the configurator could implement a nice user interface to select protocols on top of option #1 or #2, but CLI users would probably have to deal with magic protocol-indicating characters directly if we went for #2. That doesn't seem quite as nice to me as having a dedicated configuration variable to select protocols.

Another consideration under #2 is potentially having to upgrade user configuration from earlier versions to include a protocol indicator or to have whatever is reading the IR IDs being aware of what the default protocol is supposed to be if there's no special character in front. I'm not sure how these IDs work, but there might also be a potential ambiguity for IDs that may possibly contain one of the protocol-indicating characters in the first position.

DerDominik, 2017-02-14:
This change would be great to use ERLT directly via the FC.
But one suggestion: 
It is possible to configure a not-used motor output as a IR LED output?

jflyper, 2017-02-14:
@DerDominik 
> It is possible to configure a not-used motor output as a IR LED output?

It depends on if the output has a non-conflicting DMA channel/stream.
But I'm somewhat reluctant; you can find something somewhere 😉 

Ah, and you'll need an external LED current driver, of course.

paulelsner, 2017-02-22:
@DerDominik [Here](https://github.com/betaflight/betaflight/issues/112) I postet a proposal, how to use the led strip output for the ir led. Attaching the led directly worked, but of course with little energy output. I'll going to use a litte driver circuit.

DerDominik, 2017-02-27:
Hm, ok... LED_STRIP is an option, but only for those people who are not using this output
Thats why I asked for taking a non-used motor output
;-)

hydra, 2017-04-29:
hi, you should checkout cleanflight and the latest configurator.  it already now has support for iLAP and ARCiTIMER transponder systems.  There are also new MSP protocol changes.  These changes have not been picked up by the BF team yet.

hydra, 2017-04-29:
see

https://github.com/cleanflight/cleanflight/pull/2580

and

https://github.com/cleanflight/cleanflight/pull/2690

and

https://github.com/cleanflight/cleanflight-configurator/pull/450

SSPW, 2017-05-01:
When you guys are ready, i have coded a transponder implementation for the EasyRaceLapTimer protocol. 
I tested it with the SPRACINGF3EVO and RD_SSD_F3 targets on the weekend on my ERLT system. Work great!

My ERLT transponder code is here. https://github.com/SSPW/betaflight
changed drivers/transponder_ir.c and io/transponder_ir.c

I found the cleanflight transponder provider framework also. I just haven't had time to setup the CF build system yet. Kudos on your BF Docker implementation. It was Uber easy to get into BF and build it!

SSPW, 2017-05-01:
My only comment on the cleanflight implementation is the scheduling of transmissions needs to be added to the transponder config. The ERLT retransmission delay and jitter are quite different that for ilap.

hydra, 2017-05-01:
The arcitimer transponder support PR from Cleanflight *must* be merged first.

SSPW, 2017-05-01:
yes, i am aware that some from transponder provider framework like in CF has to be implemented. Just throwing out a tested implementation for an "ERLT provider" when you guys get there.

paulelsner, 2017-05-03:
Thanks, your code looks good. My implementation is very similar but I missed to share it. Thanks that you did.
Any idea how we can have multiple protocols? I also replaced the existing code, but to get this on master it would be great to have it configurable.

hydra, 2017-05-03:
yes, it needs to be configurable.  At this time the transponder changes you need are not in betaflight, i suggest you make a PR against cleanflight for this feature and then we can merge both the arcitimer and easylap into bf at the same time.

If you don't follow this approach we will end up with merge conflict situation and there will be problems with the msp protocol and problems with the gui configuration of transponder codes.

the scheduling/jitter code can be moved as you see fit as part of the work.

hydra, 2017-05-03:
@jflyper with regards to cli commands for transponder, there should not be one.

iLAP transponder codes are PRIVATE information, people pay for them.  If they were included in cli dumps, etc then people would end up accidentally posting them to the internet which is bad.

GUI configuration is all that is required.  MSP protocol in CF has already been updated to support multiple transponder providers and different length ID codes.

SSPW, 2017-05-03:
@hydra I have already started to implement a new ERLT transponder provider for CF. I just started looking at the configurator changes need on the transponder tab. Hoping to get some testing done on the weekend.

SSPW, 2017-05-04:
@hydra, i made pull requests on CF and CF-Configurator for my ERLT provider implementation. It includes protocol specific fixed delay and jitter.

mikeller, 2017-05-29:
@hydra: Just to counter your argument: If transponder codes are _not_ dumped, we will invariably end up with users dumping / restoring their config, and then come crying because they've lost their codes and have to buy new ones... (Seen so many times when people reinstall their OS, and forget to extract license keys beforehand.)

hydra, 2017-05-29:
@mikeller That has never happened in over a year of me selling flight controllers with codes.

SSPW, 2017-05-29:
couple suggestion: 

why post not a big warning, when a cli dump is done that exposes personal data (private transponder code, etc). Put private stuff at the beginning or the end so that its easy to find and edit out. Maybe even format some text around it in the cliprint, so its clearly marked. If the user posts his private data after a warning... well you cant help everyone!

or create a "public dump" private stuff is left out, and dump is suitable for posting.

or run a cli command, "public"/"private", each generates appropriate warnings. Subsequent dumps follow the mode.

Personally, i would opt first option, then i know my backups include my private stuff and i don't have to remember to do anything, and i don't have to worry about losing them. Its far more probable that a guy will lose his code because it didn't dump than a guy who will expose it publicly in a posting. I don't think i have every posted a dump, but i know i have done hundreds of dumps as backups.

SSPW, 2017-05-29:
Dump iLap code so it looks like this;

# dump
# Cleanflight / RG_SSD_F3 2.0.2 Apr 22 2017 / 14:51:15 (4328b13)

# ==========PRIVATE DATA ==========
name Quad With iLAP
transponder EDCBA9876543
# ==========PRIVATE DATA ==========

resource BEEPER 1 A08
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 C06
...

mikeller, 2017-05-29:
@SSPW: I agree with you that it's probably best if it's dumped. I was going to suggest something similar to what you've proposed with the `=== PRIVATE ===` markers. Doing it in a way that requires a special argument like `dump include_private` would be another way, but I am not sure if that's going to help, people will likely not use it when dumping / restoring.

I think today the user expectation is that `diff` / `dump` creates a complete dump, and that it can be used to dump / restore the configuration for an upgrade / reflash, with no extra steps needed.

SSPW, 2017-05-29:
In addition to dump it all and marking private, maybe add the iLaps specific warnings and/or disclaimer on the Transponder Tab of the configurator. ERLT codes and ARCtimer are limited and not private. So doesnt matter. But add text or poppup or sufficient warning at the time of code entry. Warn iLap user, that his data will be exposed during dumps. User is responsible for protecting his code and not posting it.

hydra, 2017-05-29:
It's a really bad bad bad idea to dump ilap data via the CLI! Don't do it!

It *is* however, totally fine to allow the user to backup their data via another method. e.g. backup /restore the transponder data to a binary file via the configurator.  Just do that instead and this whole privacy issue goes away.

hydra, 2017-05-29:
Also, my suggestion above does not require any new code to be added to the firmware even now.  All the information is already available via MSP.  Just add a little JavaScript to backup/restore from the transponder tab.

mikeller, 2017-05-30:
@hydra: I don't quite see why you think this is such a bad idea. There's other kinds of data that are 'private' (albeit not with financial implications), like the to be added bind information for the SPI attached FrSky RX. 

On one hand, people should not make it public. But that's already the case for so many things (like any kind of device config that includes passwords), so I think people are used to redacting their data before posting it to the internet. (Or they should learn to before they get hacked.)

On the other hand there is a user expectation that `dump` or `diff` and restore will work as a complete dump / restore, and people will lose their data if that's not the case.

ledvinap, 2017-05-30:
What about `encrypting` transponder ID in dump using CPU serial ID ? (there is entropy left even when ID is sent over MSP). Unless there is service that allows brute-forcing IDs, it should be quite safe and user friendly.

martinbudden, 2017-05-30:
@ledvinap , we'd then have to add encryption code to the ROM. F3 boards are already short of space.

ledvinap, 2017-05-30:
There is little need for strong encryption, available key space (chip ID) is small, it will be possible to try all keys. Mixing the numbers should be enough is attacker can't detect which one is legit (brute-force testing on some webservice)

hydra, 2017-05-30:
No need for encryption code, just don't dump it out via the cli in the first place.  

Not all users are good at english, some won't understand the impact of exposing their IDs. 

There's even a 'save to file' button on the CLI tab so the user might not even see any section marked with 'private' delimiters. They would just type 'dump' and hit the 'save to file' button and then upload or paste the content onto some webpage or site and expose their data without even knowing. - bad!

iLap transponder ID's are private personal information, personal information should be kept private. We have a responsibility to protect the user's data and we should respect them.  

Don't expose ilap transponder IDs via a cl dump.

mikeller, 2017-05-30:
@hydra: The same argument can be made about the IDs being paid for data, and about us having a responsibility to protect users from accidentally destroying it in a firmware update.

As a suggestion, how about we put the ID into a block of flash that does not get erased as part of a config reset? This way, they don't have to be dumped, and users will not run the risk of losing money when they click 'Flash Firmware'.

SSPW, 2017-05-31:
for the sake of moving...

1) How about dump protocols other than iLap. The others are not private or purchased. 
or 
2) Don't dump any for now. There are so few codes for the other (non iLap) protocols, most likely they need to be field changeable for each racing session. 

@hydra: is associated with the only FC vendor (SPRacing) that is actually supplying a codes with an FC. I actually can't find any others, but my search wasn't exhaustive. So, let the issue of potential data loss be the problem of the FC suppliers that supply iLap codes. Make them add a warning and disclaimers about protecting these codes. Configurator can have extra warning on iLap form.

mikeller, 2017-05-31:
I'd be for 1 for now, as it minimises the potential for users destroying their data.

rotorgeeks, 2017-06-01:
Are the iLap codes purchased separately?  If so, is it likely that any owner would still have a record of their code and be able to input it again?  So maybe not necessary to dump.

mikeller, 2017-06-01:
Ideally yes. But then how many people keep email receipts of their online purchases?

hydra, 2017-06-01:
@SSPW obviously my vote is 2.  The point about them being field changable is valid.  Less code in the firmware too. 

Each FC and transponder we sell comes with a printed QR code.  Printed QR codes are also available for purchase.   The QR code directs the user to the SP Racing website, we do not record which customers get which ID - nor can I because I sell FC's and transponders wholesale and retail, I cannot ask retailers to record each transponder code, nor would they want to.

As already stated I only support backup of the data to files using the GUI tools, but not in plain text/cli dumps.

mikeller, 2017-06-01:
So the QR code contains the iLap code, and they can use it to recover the code once they've deleted it with an update, or is it just a single use token that allows them to get a code once?

SSPW, 2017-06-01:
SPRacing sells codes that look like this;
https://oscarliang.com/ctt/uploads/2016/04/spracing-f3-EVO-FC-QR-code-transponder-ID-racing.jpg

It would seem odd to go to all the trouble of the QR based code hex code download, if it could be repeated!

Unfortunately, i can't test that assumption, because i lost my one SPRacing QR code coupon i got with an FC, before i figured out what it was! LOL

hydra, 2017-06-01:
I'm not going into the details of how it works here. But if you erase your firmware and still have the QR code you can scan it again.

DanNixon, 2017-06-01:
(maybe a little off topic)

What is the "personal" aspect to an iLap code? Is it just guaranteed to be unique or will the timing system actually not work if I just entered something random?

mikeller, 2017-06-01:
If the QR code is a permanent access token to your code, then I think the way to address your concerns about 'privacy' is to make the iLap code write only - in the CLI **and** in the GUI. There are probably more screenshots of the GUI posted than CLI dumps, and I've seen people post their JSON config files, so adding the code to either of these will cause it to be posted as well. Just add a 'use your QR code to get your code' note to the iLap entry tab.

closed 2018-07-10 by stale[bot]