betaflight/betaflight #2132: Blackbox fails to initialize formatted SD card

dimant, 2017-01-16:
SPRacing F3 Evo

1. format sd card with sd formatter
2. insert into FC
3. connect to USB
4. go to blackbox tab

expected: SD card ready
actual: Filesystem starting... (hangs there indefinitely)

works on betaflight 3.0.1
I haven't made any other changes to the config

raphaelcoeffic, 2017-01-16:
Works here on 3.1 RC8, same FC.

dimant, 2017-01-16:
I have 100% repro. Is there anything I can do to help diagnose?

Turismo05, 2017-01-16:
I have the exact same problem same FC. If I flash back to 3.0.1 everything is fine.

sinned88, 2017-01-16:
Same here. I am using a 16GB Card. 3.0.1 works

johnnyruz, 2017-01-16:
Had the exact same issue on my Betaflight F3. Setting the sdcard_dma = OFF resolved the issue for me but I'm not sure of any side effects of that setting. I get the correct files in the LOGS folder but I also see a few dozen 0KB files on the SD card with gibberish file names.

FC: Betaflight F3
SD CARD: Samsung Evo Pro 32GB (but same results with Sandisk Extreme Pro 32GB)

Turismo05, 2017-01-16:
Don't know if I should open a separate issue for this but if i have an Sdcard inserted and run Dshot, motor 1 doesn't run. Remove Sdcard and motor 1 runs fine.

mikeller, 2017-01-16:
@Turismo05: Setting `sdcard_dma = OFF` in CLI should make it work.

@johnnyruz: This is a known issue. The current implementation of DShot requires one DMA channel for every motor running DShot. Taking into account the limited number of DMA channels and constraints on their use, this severely restricts the availability of DMA channels for other things when using DShot. As such, things that could use DMA (like writing to SDCard) have to be set to not use DMA if possible to make DShot work on a lot of targets.

Turismo05, 2017-01-17:
Thanks mikeller i'll give that a go.
Is there any side affects to running with it off?

mikeller, 2017-01-17:
Probably a reduction in max logging speed, i.e. you might get logging drop outs depending on what speed you are running your loops and logging at.

closed 2017-01-19 by borisbstyle