betaflight/betaflight #2140: Cannot connect to SP Racing F3 NEO board after flashing with 3.1-RC8

rgn15996, 2017-01-16:
Since the inclusion of new [DSHOT code](https://github.com/betaflight/betaflight/commit/90733e469eca3945396369567e916331ecc0097e) for the NEO I am unable to connect to the board. This affects all Jenkins builds from build 942 and includes 3.1-RC8. Reflashing to earlier versions (or cleanflight) allows connection. This has been independently verified by another NEO user.

hydra, 2017-01-16:
fixed here:

https://github.com/betaflight/betaflight/pull/2143

rgn15996, 2017-01-16:
Tested build with @hydra PR #2143 and confirmed on the bench that can now connect and configure the FC. Also confirmed that this allows DSHOT600 on  motor1-4 outputs

closed 2017-01-16 by rgn15996