betaflight/betaflight #2142: SP Racing F3 NEO VTX Button and VTX control does not work on 3.1-RC7

rgn15996, 2017-01-16:
The NEO VTX stack has a pushbutton to control band/channel/power/state selection. This does not operate in Betaflight 3.1-RC7. VTX control with the new CLI commands does not appear to work correctly either.

hydra, 2017-01-16:
requires code merge from cleanflight.

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/vtx.c#L67-L145

raphaelcoeffic, 2017-01-17:
Somebody with a full NEO stack? I would do it, but did not plan on buying it ;-)

rgn15996, 2017-01-17:
More than willing to help with any testing/feedback - I'm tech savvy but not familiar with this codebase in any depth and it's about 25 years since I did any serious C programming!

rgn15996, 2017-01-18:
While this is now framed as a feature request for making the VTX button function, I think that there is also a bug in the current implementation because I'm unable to select the VTX channel/band from either the CLI or via RC stick commands. I'm prepared to accept that this is user error as I'm only just starting to look at the src for betaflight and I really don't understand the control flow yet, but @hydra could you either verify this or point me on the right path for VTX control of the board in betaflight?

hydra, 2017-02-05:
@rgn15996 no, I'm not familiar with the code, just look through the vtx source files and you should be able to figure it out, it's where I'd start 😄

hydra, 2017-03-05:
once @jflyper has completed the VTX API work I can take a look at the NEO specifics but there's no point me starting until he's completed that work otherwise it will just be merge-conflict hell.

mikeller, 2018-02-01:
@rgn15996: Has this been solved in the meantime?

mikeller, 2018-02-11:
Closed with no response.

closed 2018-02-11 by mikeller