betaflight/betaflight #2147: Inconsistent cycle time with blackbox enabled, 8k pid loop.

hyonlim, 2017-01-17:
Hi,
 I am using custom made, F4 board.
 CPU load is pretty low when 8k/8k setting.
But, when I enable blackbox, the cpu cycletime is very inconsisstent.
 I've tried 1k blackbox setting, still inconsistent cycle time.
 What about another board? have you tried to log blackbox with 8k/8k setup?

borisbstyle, 2017-01-17:
Sorry I will close this one as 0 information is given and all existing targets do handle this well as far as I know.

closed 2017-01-17 by borisbstyle

hyonlim, 2017-01-17:
Hi, I found some clue on this problem.
Our board, support both SPI flash and SD card, this happens only when SD card is *not* inserted.
We are debugging on this problem too. Hopefully, we can revert our debug result to the repository.
In the meantime, could you please take a look, to see if any SD card insertion check cause this problem?

-Hyon

hyonlim, 2017-01-17:
Also, I wonder that if any board in the market has SD card and SPI flash onboard.

-Hyon

mikeller, 2017-01-17:
@hyonlim: BLUEJAYF4 has onboard SPI flash and SD card. And there have been some problems with logging to SD card with this target in the past.