betaflight/betaflight #2156: Blackbox logging for sensors only (without arming).

ckwkalk, 2017-01-18:
Hi, It doesn't seem possible to enable Blackbox logging without arming the system? I want to test sensor performance on the bench with just a battery connected (i.e. untethered). At the moment it seems that there is no way to enable logging without arming and no way to to auto-arm to allow sensor testing. I am aware that you can carry out tethered logging but this is not a solution if testing multiple boards simultaneously.

mikeller, 2017-01-18:
Yes you can, with a little workaround: If you enable `blackbox_on_motor_test` in CLI, blackbox logging will be enabled when unarmed and at least one of the motors is not set to be stopped.

ckwkalk, 2017-01-18:
@mikeller Sounds promising! Will give it a try and report.

ckwkalk, 2017-01-19:
@mikeller Nearly there! So  `blackbox_on_motor_test` does enable logging without arming but I didn't manage the second part you suggested - "set one of the motors not to stop". I can go in through the GUI and manually start in Motor Test Mode but I couldn't see a way to do this that would be saved in the CLI  through a power cycle (i.e. letting me untether it and connect to battery). I have tried to adjust the `min_command` and `motor_stop` feature but with no success.

closed 2017-01-19 by borisbstyle

mikeller, 2017-01-19:
@ckwkalk Ah, yes, you can not configure the firmware to automatically spin up the motors on startup (which is a good thing from a safety perspective). Are there any negative implications if you conduct your tests while the craft is plugged in by USB?

ckwkalk, 2017-01-20:
@mikeller well if you imagine that I want to stack three boards, perform some motions and then compare their outputs, tethering would mean needing three computers! I realise this is a non-standard use-case. So as you put it what I really want is a way to "automatically spin up the motors on startup" which the firmware really doesn't want to permit (I hadn't appreciated how much the firmware is doing before attempting to get around it!). Thanks again. C.

mikeller, 2017-01-20:
Hehe, that IS an out there use case. But then, presumably, you should be able to build your own custom build for this experiment. Easiest is probably to rig https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L898 to always return true.

ckwkalk, 2017-01-20:
@mikeller well I've had a go but not succeeded - I think that because it is looking for the MSP_SET_MOTOR command and DSHOT_DISARM_COMMAND it is still necessary to do this through the GUI/USB? I tried to set the #L298 static BlackboxState to RUNNING which seemed to work but caused it to write corrupted data (similar to issue  [#611](https://github.com/betaflight/betaflight/issues/611))