betaflight/betaflight #2157: Can't connect with USB with DOGE after reflash

brycedjohnson, 2017-01-19:
Failed with RC10. 
Need to push boot pin to reflash.  Tried with erase all and still failed.

Did a little search to see where it brok, was previously running a older RC from a week or two ago.  Almost narrowed it down - RC9 worked (jenkins build 959).  Tried Jenkins 966 and it was busted again.  Didn't have time right now to try all the versions inbetween.

brycedjohnson, 2017-01-19:
Might be this commit?  0ebff695843583298e080734e96af279cf5aedc7

AndersHoglund, 2017-01-19:
Re-tested my two F3 VCP boards, SPRF3EVO and BFF3, with RC10. Works fine. Also CC3D and CC3D_OPBL still works. Unfortunately I do not have any DOGE to verify.  Could you please do some tests on build 962 and 963, and preferably also tests on commit  0ebff69 and the commit prior to that 1e80ff1c5b5b59f1d050833adb5c265a40e41fa6 , verifying that it is exactly this commit that kills DOGE.
We could refine the USB VCP interrupt fix in PR #2153 to be valid only for F1's. And maybe also only for OPBL targets.  Different STM bootloaders in F1 and F3.
What other F1 and F3 VCP targets are there to be tested?

borisbstyle, 2017-01-19:
@brycedjohnson When you flash back and fourth between RC9 and RC10 does it work?

Sometimes you have to reflash another version to reset the config.

Also sometimes PC needs a reboot.

AndersHoglund, 2017-01-19:
@brycedjohnson There are MSP version changes in this area also, you might need the latest BFC.

brycedjohnson, 2017-01-19:
Yes swapping back to RC9 works.  @borisbstyle. Haven't tried a reboot yet.  I'll do build 962 and 963 first.  

Running configurator 1.8.9.   @AndersHoglund

brycedjohnson, 2017-01-19:
Also I don't have the build toolchain set up, so I can't test the separate commits.

AndersHoglund, 2017-01-19:
There are many different BFC 1.8.9 ;-) , you need to download and install current github HEAD.

AndersHoglund, 2017-01-19:
OK, I have an un-commited experimental PR branch with the Interrupt Fix only for STM32F10X, please find binaries attached. 
[betaflight_3.1.0-RC10++_Experimental.zip](https://github.com/betaflight/betaflight/files/716710/betaflight_3.1.0-RC10._Experimental.zip)

brycedjohnson, 2017-01-19:
Build 962 worked and 963 didn't work.  When it doesn't work I get (in windows): unknown USB device (Device Descriptor Request Failed)

[image]

I'll try your attached one quick - then I gotta run to work!

brycedjohnson, 2017-01-19:
@AndersHoglund Your experimental build worked for me!

Let me know if you want me to test anything else for the DOGE.

AndersHoglund, 2017-01-19:
OK, thnx. But this is weird, I really do not understand what could be different with DOGE USB VCP compared to BFF3 and SPRF3. But never mind, I will commit and issue a new PR with this fix.

borisbstyle, 2017-01-19:
@brycedjohnson did you reboot the PC in between?

yesterday during some testing I experienced something simillar with LUXV2_RACE, but had to reboot the PC to make it work

brycedjohnson, 2017-01-19:
I didn't try that yet.  I'll give it a try when I get back home tonight.

brycedjohnson, 2017-01-20:
@borisbstyle Restart and it didn't work.  Windows didn't even detected the USB device after restart with the device descriptor error I had before.  Boot button and using the interrupt fix from @AndersHoglund and it worked great again.

borisbstyle, 2017-01-21:
Funny thing is that DOGE does work for me!

brycedjohnson, 2017-01-21:
That is interesting, maybe I'll try another cable!?

closed 2017-03-26 by brycedjohnson