betaflight/betaflight #2162: SPRACINGF3EVO DShot 600 - Motor 1 does not start

JBlocks, 2017-01-19:
**Setup:**
- Seriously Pro Racing F3 Evo
- Betaflight 3.1.0-RC10
- DShot 600
- Remapped motor 4 to 5 and plugged/wired to pin 5.
- All 4 motors are shown in DMA section of resource list.

**Issue**
 - Motor 1 will not start

----
I followed up the wiki "ESC DSHOT" for the SPRACINGF3EVO and soldering the motor 4 to the motor 5 pin and remapped it to motor 4 A06. The motor works fine on arming. 

In motor tab of betaflight GUI with DShot 300 & 600:
I can start motor 2,3,4 (4 is remapped to 5), but motor 1 will not start by increase the speed manually.
Motor 1 does not start with the DShot 300 and 600 settings. With OneShot, all 4 motors are running.
LED_Strip and Transponder are disabled.

**Wiki introductions for SPRACINGF3EVO**
> SPRACINGF3EVO - Must move MOTOR 4 to new pin assignment (CLI = resource MOTOR 4 A06). Then solder ESC for motor #4 to motor output #5, fixes DMA conflict with motor outputs 2 and 4.
> Solder a wire from Output #4 header pin to Output #5 header pin. Follow above and to re-map output type in CLI:
> resource motor 5 none
> resource motor 4 A06
> save

What am I doing wrong?

rgn15996, 2017-01-19:
Just a suggestion - have you looked at the output of resource and resource list CLI commands to see if there is a DMA conflict with a feature on the board - maybe BLACKBOX or LED strip? I'm feeling my way with this as I know very little about embedded systems and the F3 chip, but it may be that you need to disable one of those features. I'll follow with interest as I have an EVO build that I will want to convert to run DSHOT soon.

raphaelcoeffic, 2017-01-19:
Which ESCs are you using? Did you remove the signal capacitors? Have you tried to swap ESC+motor, for example 1 & 2, just to be sure this is not related to the ESC/motor?

JBlocks, 2017-01-19:
Using DYS XSD 30A with DShot 300/600 support and newest BLHeli firmware on each ESC.
See http://www.dys.hk/ProductShow.asp?ID=243

Yes, I swap ESC & motor and tested motor at bottom right with pin header for motor 1. Same behaviour like in the first post.

raphaelcoeffic, 2017-01-19:
So it's always the motor connected with output 1 that does not activate, correct? (sorry, having trouble to understand you properly).

JBlocks, 2017-01-19:
Yes, it is always connected with output 1.  - See attachment. 
The blank output is motor 4 and remapped to output 5. - As mentioned in the article ESC DShot (wiki).

[image]

wind0r, 2017-01-19:
Can't confirm this bug on both my SP3EVO quads

@JBlocks can you post your cli diff here please? :)

```# version
# Betaflight / SPRACINGF3EVO 3.1.0 Jan 19 2017 / 00:58:23 (8f017b9)

# resources
resource MOTOR 4 A06
resource MOTOR 5 NONE

# master
set motor_pwm_protocol = DSHOT600
```

borisbstyle, 2017-01-19:
Did you verify that esc on that motor actually supports dshot?

Swap esc connections to verify

JBlocks, 2017-01-19:
Thanks for all replies and hints.
After checking all the cli-settings I found the solution. I can toggle the "blackbox" settings in the gui and save the settings, but the property will not be set after reboot.

If I disable the blackbox via the CLI and save, then the setting is adopted and all 4 motors run with DSHOT 600. There seems to be a DMA conflict with the blackbox.

**Resource and resource list after disabling the blackbox**
> # resource
> resource BEEPER C15
> resource MOTOR 1 A00
> resource MOTOR 2 A01
> resource MOTOR 3 A02
> resource MOTOR 4 A06
> resource MOTOR 6 A07
> resource MOTOR 7 B00
> resource MOTOR 8 B01
> resource MOTOR 9 B10
> resource MOTOR 10 B11
> resource PPM A15
> resource LED_STRIP A08

> # resource list
> Currently active IO resource assignments:
> (reboot to update)
> --------------------
> A00: MOTOR 1
> A01: MOTOR 2
> A02: MOTOR 3
> A03: FREE 
> A04: ADC_BATT 
> A05: ADC_CURR 
> A06: MOTOR 4
> A07: FREE 
> A08: TRANSPONDER 
> A09: SERIAL_TX 1
> A10: SERIAL_RX 1
> A11: FREE 
> A12: FREE 
> A13: FREE 
> A14: SERIAL_TX 2
> A15: SERIAL_RX 2
> B00: FREE 
> B01: FREE 
> B02: ADC_RSSI 
> B03: SPI_SCK 1
> B04: SPI_MISO 1
> B05: SPI_MOSI 1
> B06: I2C_SCL 1
> B07: I2C_SDA 1
> B08: LED 1
> B09: MPU_CS 
> B10: FREE 
> B11: FREE 
> B12: SDCARD_CS 
> B13: SPI_SCK 2
> B14: SPI_MISO 2
> B15: SPI_MOSI 2
> C13: MPU_EXTI 
> C14: SDCARD_DETECT 
> C15: BEEPER 
> F00: FREE 
> F01: FREE 
> F04: FREE 
> 
> 
> Currently active DMA:
> --------------------
> DMA1 Channel 1: MOTOR 3
> DMA1 Channel 2: TRANSPONDER
> DMA1 Channel 3: FREE
> DMA1 Channel 4: FREE
> DMA1 Channel 5: MOTOR 1
> DMA1 Channel 6: MOTOR 4
> DMA1 Channel 7: MOTOR 2
> DMA2 Channel 1: ADC
> DMA2 Channel 2: FREE
> DMA2 Channel 3: FREE
> DMA2 Channel 4: FREE
> DMA2 Channel 5: FREE
> 

Now unfortunately the problem arises that the motors automatically turn up, if I short-term move the throttle stick on the remote control and back to the zero position (1000). I must immediately activate "disarm", so the motors do not turn up to 100%.

- BetaFlight 3.1.0RC10
- SPRACINGF3EVO
- DShot 600
- DYS XSD 30A 300/600 DSHOT
- Throttle range 1000-200
- Arm / Disarm with Aux-Channel on the remote control
- Deactivated blackbox for the engine 1 to run

Do I have to set a configuration so that the motors do not turn up?

wind0r, 2017-01-19:
Spinning up motors on quads without props are normal!
Just watch: https://www.youtube.com/watch?v=xBaefeQtA1c

JBlocks, 2017-01-19:
Thanks!

Regarding the Blackbox, this should perhaps be included in the wiki. Or is there a workaround or setting to use the blackbox in combanition with dshot?

wind0r, 2017-01-19:
If you want to use blackbox you can test cli `set sdcard_dma = OFF`
Currently i do not have a sd card around to test or confirm this.

JBlocks, 2017-01-19:
I have just done the test with the setting "set sdcard_dma = OFF". All 4 motors are running with DShot 600 and enabled blackbox feature. The blackbox data is written to the memory card and can be read  with the blackbox explorer.

You can include this information in the Wiki for the SPRACINGF3EVO FC.

**My current settings with working DShot 600 and blackbox (SDCard) feature**
> # Betaflight / SPRACINGF3EVO 3.1.0 Jan 19 2017 / 00:58:23 (8f017b9)
> 
> # resources
> resource MOTOR 5 NONE
> resource MOTOR 4 A06
> 
> # master
> set motor_pwm_protocol = DSHOT600
> 
> # disable dma for sdcard, if DShot is enabled on SPRACINGF3EVO
> set sdcard_dma = OFF

closed 2017-01-25 by JBlocks

waltr1, 2017-01-31:
Dshot page of Wiki for SPF3EVO edited:
Added link this this github page and the "set sdcard_dma = OFF" CLI command.

mikeller, 2017-02-01:
Thanks @waltr1.