betaflight/betaflight #2165: Sensors (barometer, magnetometer) not detected

christianrauch, 2017-01-19:
Hi,

The barometer (`MS5611`) and magnetometer (`HMC5883`) on my naze32 full does not get detected by betaflight (3.1.0-RC10).
The current Cleanflight version 1.14.1 does detect all sensors according to the CLI:
```
# status
System Uptime: 19 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.19
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611, MAG=HMC5883
PIDd: 1139, GYROd: 1142, I2C Errors: 0, registry size: 53
```

Only the `MPU6500` gets detected by betaflight and hence only the gyroscope and accelerator values are available.

jflyper, 2017-01-19:
Did you turn on the baro and mag in the configurator?

wind0r, 2017-01-19:
since f1 have not enough space for all betaflight features mag got disabled.
baro should be there. i think you maybe have it disabled in the configurator.

a diff or dump log would be nice!

christianrauch, 2017-01-19:
I didn't know that you can actually dis-/enable sensors. Do you mean the toggle box in the sensor tab? It is not possible to activate them.
The `MSP_STATUS` message also reports that only the accelerometer is present.

My configuration is basically the default configuration after flashing: [betaflight_dump.txt](https://github.com/betaflight/betaflight/files/717431/betaflight_dump.txt)

andre92, 2017-01-20:
Enabling sensors is not in the sensor tab but in the configuration tab.
Or type set baro_hardware = auto and set mag_hardware = auto in cli.

gerse70, 2017-01-20:
Das ist bei mir auch so. Wo muss man was einstellen?
[image]

gerse70, 2017-01-20:
bei der Software 2.9.1 sind Baro und Mag noch da.

wenn ich in die CLI gehe
-dann Dunp eingebe
- dann set mag_hardware = auto eingebe
-dann save eingebe kommt der mag auch nicht.

SoulPictures, 2017-01-20:
@gerse70 wie du Jungs oben schrieben ist der Magnetometer bei F1 rausgeflogen da kein Platz mehr im Speicher war.

As the guys already wrote, mag got disabled due to size problems

gerse70, 2017-01-20:
Ich habe eine Naze32 Rev 6 Full sollte doch der Speicher ausreichen.
Mit der Software 2.9.1. geht es auch.
Kann man den Speicher auslesen, viel frei ist?

wind0r, 2017-01-20:
naze32 rev6 is an f1 board (repeated as above: since f1 have not enough space for all betaflight features mag got disabled.)

if mag and all other features are enabled the linker will stop creating the binary since it is so big.
you can create yourself a custom binary with other features disabled and mag enabled

gerse70, 2017-01-20:
Welchen FC können Sie empfehlen

mikeller, 2017-01-20:
Hallo @gerse70: Ich wuerde einen mit F4 Prozessor empfehlen, da diese Generation im Moment noch viele Reserven im Bezug auf Speicherplatz und Rechenleistung hat, zum Beispiel den hier:

https://github.com/betaflight/betaflight/wiki/Board---BLUEJAYF4

christianrauch, 2017-01-21:
I finally enabled the barometer in the configuration tab.
Is the magnetometer completely removed in the F1 code? Or are there `#define` do enable/disable features at compile/linking time? I somehow imagine that reading the magnetometer values and use them to get a frequent and accurate absolute heading consumes less than the GPS driver and functionality.

wind0r, 2017-01-21:
you can reenable mag feature but you need to disable other stuff then https://github.com/betaflight/betaflight/blob/master/src/main/target/NAZE/target.h#L108-L112

for my new quads i have beatflightf3 and bluejayf4 around, both are great :)

christianrauch, 2017-01-21:
Enabling the magnetometer in `src/main/target/NAZE/target.h` lines 108-110 and disabling the LED strip feature in line 149 (comment `#define LED_STRIP`) reduces the binary size such that it fits on the F1. The heading is then correctly reported as the angle to the magnetic north.

Is there a difference between the drivers for the `HMC5883` and `HMC5983`?
The `target.h` mentions `USE_MAG_HMC5883` but the naze32 rev6 full actually has a `HMC5983` which provides more accurate value than the `HMC5983`.

jflyper, 2017-01-21:
> Is there a difference between the drivers for the HMC5883 and HMC5983?
The target.h mentions USE_MAG_HMC5883 but the naze32 rev6 full actually has a HMC5983 which provides more accurate value than the HMC5983.

HMC5983 seems to be fully upper compatible with HMC5883L.

Only difference is the presence of API for internal thermometer.

closed 2017-01-23 by borisbstyle

nipunnawanjana1, 2018-02-20:
I have work this https://youtu.be/9Es1NzxLyWE