betaflight/betaflight #2166: Revert back some OSD functionality.

marbalon, 2017-01-19:
I know there is a huge rework inside OSD/CMS but two features was quite useful.

- LED strip color change - it is really often used feature byt many pilots during races
- less important but useful was moving elements using sticks

Is there any reason to not revert this functionality back to OSD menu ?

borisbstyle, 2017-01-26:
I agree about this one. Yesterday I wanted to move it and found out if was removed

@jflyper Any ideas why it was removed

I also think it should be nice to add blinking battery warning in the middle of the screen or at least have it configurable?

jflyper, 2017-01-26:
@marbalon @borisbstyle I remember I dropped it ...

jflyper, 2017-01-27:
Okay, I found the reason for led strip color feature being dropped : [comment](https://github.com/betaflight/betaflight/pull/1422#issuecomment-258635758)

jflyper, 2017-01-27:
LED color is ready for testing. I can't test it until next mid next week.

[CMS] Revert ledstrip color function #2232

borisbstyle, 2017-01-31:
added in 3.1.1

closed 2017-01-31 by borisbstyle