betaflight/betaflight #2171: Fury F3- updated all, motors 1+4 continuous spool up

gibletfritter, 2017-01-20:
Hello, I have a fury f3 runing 8k/4k Dshot600 and escs that are updated and capable. Upon installing the newest hex for my board, the quad now will allow a normal-operating range to about 40% throttle. Anything past 40%- if i reduce throttle, motors 1+4 start to slowly spool up and will keep increasing, regardless of throttle value. This only happens if i exceed 40%. My connections are good and the reciever/tx are functioning as they should. These problems only arose post BF3.1.

gibletfritter, 2017-01-20:
Also, I have replaced the ESCs with brand new ones that are very recently developed and up to date. And I reflashed everything. The issue still persists.

wind0r, 2017-01-20:
Copy+Past:
Spinning up motors on quads without props are normal!
Just watch: https://www.youtube.com/watch?v=xBaefeQtA1c

gibletfritter, 2017-01-20:
Thank you, headache is now gone haha. I've even seen that video
before....Guess I was way over analyzing, its got to be the problem.

wind0r, 2017-01-20:
No Problem :) It would be nice if you close the issue now :)

gibletfritter, 2017-01-20:
I'll remove the post...It seems flyable, but if I exceed 40% throttle, the
issue is still present. I figured that since it's motor 1+4 that are more
active, the PIDs are trying to provide more yaw motion. 2+3 never totally
spin-down. Any ideas why?

closed 2017-01-20 by gibletfritter