betaflight/betaflight #2172: PWM buzzer with CC3D

mikenorgate, 2017-01-20:
I have one of these power distribution boards (https://hobbyking.com/en_us/led-power-hub-5in1.html) which has a build in lost model buzzer. The buzzer is designed to be activated using on of the AUX channels, however as I'm using PPM it needs to be connected to the FC.

I've read a number of forum and blog posts that say this can be achieved by using either SERVO_TILT or CHANNEL_FORWARDING, however I can't seem to get any of them to work.

I know on the CC3D board pin 6 on the RC_Ouput is not used, is there any way to enable this pin on an AUX channel to run this buzzer?

mikeller, 2017-01-21:
With 3.1, no pins are set up as servo outputs by default. If you want to use motor output 6 as a servo PWM output, you need to reconfigure it in CLI as a servo output: `resource motor 6 none` and `resource servo 1 pa2`. Then, after a reboot, `resource list` should contain `PA6: SERVO 1` and you should be good to go to set up `SERVO_TILT` or `CHANNEL_FORWARDING`.

mikenorgate, 2017-01-21:
I get an error when trying to assign to `pa2` as it does not exist.

Here is the output from `resource list`

> A00: FREE 
A01: PPM 
A02: FREE 
A03: MPU_EXTI 
A04: MPU_CS 
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE 
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: FREE 
B00: FREE 
B01: FREE 
B02: INVERTER 1
B03: LED 1
B04: FREE 
B05: FREE 
B06: FREE 
B07: MOTOR 3
B08: MOTOR 2
B09: MOTOR 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: FLASH_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C14: FREE 
>
>Currently active DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE

Also is it possible to change the beeper mode to use the servo output? So that other functionality that uses the beeper would work

mikeller, 2017-01-21:
Sorry, my bad, it's late here. The second command should be `resource servo 1 a02`.

AndersHoglund, 2017-01-22:
Got a similar Q from a guy using SERVO_TILT and motor 1 output to a PWM beeper.

I tried to use the same technique as above to re-assign MOTOR 1 and 2 as SERVO 1 and 2. This however killed all motor outputs in the motor test tab. The drawbars will not even move. Also, the ESC does not initialise and arm any more. I have only one ESC to test with, connect to motor 3 output. Servo on motor1 does not move either. Tested on NAZE32.

So, big Q: Is SERVO_TILT completely broken now?

Will try CHANNEL_FORWARDING as of above instead...

AndersHoglund, 2017-01-22:
No luck with CHANNEL_FORWARDING either. Erratic ESC behaviour when throttling in motor test, ESC restarts. With this NAZE32 resource map:

    # version
    # Betaflight / NAZE 3.1.0 Jan 21 2017 / 14:44:45 (951e827)
    
    # resource
    resource BEEPER A12
    resource MOTOR 1 A08
    resource MOTOR 2 A11
    resource MOTOR 3 B06
    resource MOTOR 4 B07
    resource SERVO 1 B08
    resource SERVO 2 B09
    resource PPM A00
    resource PWM 1 A00
    resource PWM 2 A01
    resource PWM 3 A02
    resource PWM 4 A03
    resource PWM 5 A06
    resource PWM 6 A07
    resource PWM 7 B00
    resource PWM 8 B01
    resource LED_STRIP A06

@mikenorgate did you manage to get this working on CC3D?

mikenorgate, 2017-01-22:
I've been unable to test it yet due to erratic motor behaviour in the latest version. 

Upon arming motor 3 spins to max throttle and 1 and 4 go to half throttle. 

So I'm back on version 3.0

mikeller, 2017-01-22:
@AndersHoglund:

> I tried to use the same technique as above to re-assign MOTOR 1 and 2 as SERVO 1 and 2.

Does this mean he removed resources for motor 1 and motor 2, and didn't just move them 'up'? This won't work since resources don't support sparse lists, i.e. if motor 1 isn't configured, it won't even try motors 2, 3, 4.

Another thing to consider is that pins used for servos need to be on a different timer than pins used for OneShot / Multishot. On NAZE, motor 1 and 2 are on timer 1, motors 3 to 6 on timer 4. So to get the above config to work, servos need to be on A08 and A11, motor 1 to 4 on B06 to B09.

AndersHoglund, 2017-01-23:
Well, the SERVO_TILT feature used to shift motors two steps up in previous versions. Why? I have no idea, probably for some resource conflict reason. Is this not done automatically any more and the user is supposed to shift this up using resource re-mapping? How this is conceived to be working with the new IO and resource allocation system beats me. Mr Resource Manager will have to answer that. All those tricky IO/timer/DMA/Irq allocation details can not be left to the end user to figure out. 
You tell me how to use SERVO_TILT and CHANNEL_FORWARDING. On second thought, don't tell me. Tell the users. And if all this different per target,  we are in deep sh.. documentation-wise.

Then instead, we could take a step back and take a look at what we have here. Old inherited features like this, are they still valid and useful on a quad racer? Should we keep supporting old stuff like this, with all it's implications and complications, or just simply scrap them? Remove this croft I would say. When taking ROM-space into account it makes even more sense to remove them. OK, a tri-copter will need servo support, but camera tilt? And channel fwd? Who uses that? For what? Should we keep those features just for the sake of a dumb PWM-beeper? What other use cases are there? I mean valid use cases and really strong requirements. 
I told my friend with the original problem to ditch the PWM-beeper and use a simple dumb beeper on the existing dedicated pin on the Naze. Simple solution to an overcomplicated problem, or a non-problem really.

romangorchakov, 2017-10-01:
Hey guys just in case it's not solved yet -- I'm so happy I could get it working because I googled and got nthng. CC3D with BF 3.2.0 RC6, Matek beeper's "B-" (not "battery", "beeper") connected to Motor 6. Then "resource motor 6 none" (just in case) and "resource beeper a02" and that's it! Whew!! )

MrMonsieur, 2017-10-31:
@mikeller 

Thank you so much. I've been reassigning the servos and just resetting the naze and betaflight because I just could not get my stupid buzzer on the matek pdb v3.1 to work.

This is just information gold: 
»Another thing to consider is that pins used for servos need to be on a different timer than pins used for OneShot / Multishot. 

On NAZE, motor 1 and 2 are on timer 1, motors 3 to 6 on timer 4. So to get the above config to work, servos need to be on A08 and A11, motor 1 to 4 on B06 to B09.«

closed 2018-04-21 by DieHertz