betaflight/betaflight #2173: BluJay F4 type FC (BG F4 V3 AIO Board (PDB, OSD) with ICM20608 gyro) - no USB connect

RipperDrone, 2017-01-20:
Anybody managed to get this board to connect over USB?

Bootpins shorted throws it into DFU mode as supposed, and flashing BluJay F4 BF firmware completes successfully (full erase), but after finishing the flash the board cannot connect to USB.

Tried both BF 3.0.1 and 3.1RC10 (plus today's Jenkins Build) - no connection to USB. Tried the usual drivers game (VCP, Zadig, CP210...) to no avail.

Any ideas? On BG forums, some users seem to have it running with BF. Latest changes have affected USB protocols, so I wonder if this has something to do with it...

mikeller, 2017-01-20:
If not in DFU mode, is the board recognised by your operating system, and does the corresponding port appear in the configurator when connected?

RipperDrone, 2017-01-20:
No, only seen with a port shown in BFC when in DFU mode, dead otherwise. No documentation came along with the FC, but I can see red and green LEDs on FC flashing several times in a sequence, then pause, then flash the sequence again. On BlueJay I found this means firmware error/problem. Maybe the firmware is not matching the FC, despite BG and users mentioning it as BF3.1/3.0 BlueJay F4 compatible?

RipperDrone, 2017-01-20:
firmware flash completed with std green message successfully flashed, shown in log as well. It only does not connect to USB after flashing, no port shown... My Win10 PC sounds the 'USB diconnect' sound when I plug in the FC. Strange...

mikeller, 2017-01-20:
Sounds very much like a driver problem. (DFU  and non-DFU mode present themselves as different devices, so require different drivers.) Check in the device manager for problems, uninstall / reinstall the driver.

RipperDrone, 2017-01-20:
Contrary to the BlueJay F4, this board does not have an SD card slot. Any idea if the BlueJay F4 firmware is still the correct one to flash? 

Here's the BG reference to the board I have: http://www.banggood.com/STM32F4-F4-V3_0-Racing-Flight-Control-with-BECPDBOSDXT60-for-Multirotor-p-1109743.html?p=RU2509087943201210XZ

AndersHoglund, 2017-01-20:
For a VCP/DFU board like this you only need 1. WinUSB driver when in bootloader. Zadig or ImpulseRC driver fixer will set that up for you. 2. Normal COM port mode requires STM VCP drivers. 
That is all. Don't mess with Silab CP210x drivers now, not applicable with those boards. Will only be needed when you actually have a FC with the Silabs CP210x USC-device.  
If you can connect an AirBotF4, both as DFU and COM port device, then your driver installation is OK. Broken FC I would say. Or not BJF4 compatible. My BJF4 with BF3.1-RC10 connects just fine.

RipperDrone, 2017-01-21:
@AndersHoglund I see. Still there's a hint in the logs maybe: 

Flashing without full chip erase logs 1024kiB of flash detected, but only 384kB of flash erased.

Flashing WITH full chip erase runs the full erase step up to 100% progress bar, then freezes the progress bar there, displaying Flashing... as text info on the bar and nothing else happens.

Could it be that flash memory & firmware somehow don't match up in size?

mikeller, 2017-01-21:
No, that's entirely normal. When flashing without full chip erase, it will only erase the space that it needs for the new firmware.

And there is a bug with full chip erase that leads to the behaviour that you describe.

RipperDrone, 2017-01-21:
Ok, I'll give up to fix it myself and claim hardware warranty then... thanks a lot!

closed 2017-01-21 by RipperDrone

AndersHoglund, 2017-01-21:
The Full Chip Erase Bug: https://github.com/betaflight/betaflight-configurator/issues/200

AndersHoglund, 2017-01-21:
Also, if the board can be flashed OK over USB then the hardware is OK. I would say a driver issue, but then you could connect ABF4 that uses the same drivers....
Aaah, this is a classic gotcha: Did you remove to boot pad bridge completely? Some invisible strands remaining can give this behaviour. Clean up and inspect.

RipperDrone, 2017-01-21:
Cleaned it up, inspected it. When I plug in USB, blue LED only lights when pads are shorted (I do it with a pair of pincers, pads are not even tin plated/covered). Opening the pads connection again, USB connect triggers red and green lights, followed by error signal (4x short + 2x long red LED flashes)

RipperDrone, 2017-01-21:
I saw DOGE has similar problems (no USB connect after DFU flash) - seems there is a way to try recover by FTDI adapter serial connection and flash over UART? Not sure this holds for the F4 as well...

AndersHoglund, 2017-01-21:
Sure, you can probably use UART1 for flashing on F4, only tested that on F3s. But flashing is not the problem. USB VCP seems to be the issue. Could this be related to the same problems we had on CC3D, but the fix for that killed (some) DOGE. I have not seen any USB VCP problems on my F3s or F4s, with and without the CC3D vcp interrupt fix. Ref #2163  and #2153

RipperDrone, 2017-01-21:
Tried to flash, but connection could not be made via RX1 and TX1 pins to an FTDI/USB adapter - it only opens for a few seconds, then closes again.

Any recommendation which Jenkins Build I should try moving back to in order to verify whether this is to do with the CC3D related issues of not being able to USB connect? (When did all the VCP/USB/DFU interrupt/non-interrupt struggle start?)

RipperDrone, 2017-01-21:
And which version of configurator will I have to use then?

RipperDrone, 2017-01-21:
Just tried 3.0.1 with BFC 1.8.5 - same :-/

RipperDrone, 2017-01-21:
When connecting VCP over FTDI, BFC log says: 
2017-01-21 @ 15:17:47 -- No configuration received within 10 seconds, communication failed

AndersHoglund, 2017-01-21:
FTDI is not VCP..... Did you try flashing or connect for configuration?  For Bf3.1 config you would have needed to enable MSP on UART1, but that is not possible since you would have to connect via USB first to do that. MSP on UART1 is off by default now.
In BF3.0.1 MSP on UART1 should be on by default.

RipperDrone, 2017-01-21:
HAve tried both: Connecting over FTDI adapter for flashing and for config. Message came up when connecting for config. For flashing, it would not connect at all / seems to be DFU-only...

RipperDrone, 2017-01-21:
Will flash BF3.0.1 now and retry FTDI connect for config...

RipperDrone, 2017-01-21:
:-(

2017-01-21 @ 15:54:32 -- Unrecoverable failure of serial connection, disconnecting...
2017-01-21 @ 15:54:32 -- Serial port successfully closed

AndersHoglund, 2017-01-21:
Did a test to flash my Revo (F4) over UART1 , failed. Timeout.
But according to the ST app note it should work: http://www.st.com/resource/en/application_note/cd00167594.pdf page 23.
Strange, but never mind; Flashing is not the issue.