betaflight/betaflight #2178: Can't get smartaudio working on softserial

u8mydog, 2017-01-22:
Can anyone verify that smartaudio works using softserial?

I have:
tbs unify pro hv with white audio wire connected to softserial1 TX
Using F3S AIO v1.0 controller with integrated mwosd
uart1=msp
uart2=S.Port telemetry
uart3=SBUS serial RX

In CMS menus, vtx status shows all values 0 (except baud and sent). I also do not see the configuration string. I see '- -- ---- ---'. So it appears the softserial1 is not receiving.

u8mydog, 2017-01-22:
btw: Im using betaflight 3.1.0-RC10 and scarob mwosd 1.6.6.3.

jflyper, 2017-01-22:
Smartaudio is not compatible with softserial atm, due to it's special requirement on the line characteristics.

u8mydog, 2017-01-22:
Smartaudio is a single wire tristate interface at a slow baud rate. I would have thought that softserial could support tristateing tx while waiting for a response. Especially since smartaudio does not talk until spoken to.

3 physical uarts disappear quickly when using OSD, SBUS and S.Port. Even bit bashing, instead of softserial, for smartaudio, would be beneficial.

jflyper, 2017-01-23:
For post-3.1, we are looking at

1. One-wire software serial (timer equipped pin).

2. One-wire big-banging (any pin).

Hang on.

geofrancis, 2017-01-26:
Have you got a guestimate Eta on softserial smartaudio?

jflyper, 2017-01-27:
@geofrancis 
Not really sure. Softserial with mono-timer started to work last night, and I'm now working on single wire half-duplex case. So, say, 2 weeks tops for dev. I'm not sure about release schedule of 3.1.1 or 3.2.

kalosjann, 2017-02-14:
I got the same issue with a similar configuration.
Fc: Holybro Kakute AIO. 3 serial ports occupied by usb, osd, sbus.
Vtx: tvs unify pro hv
I reassigned the PWM 6 to the PPM pin (the only free pin on this board) and enabled the softserial on this port at 19200.
After that I connected the smartaudio port to the ppm pin.
Now smartaudio seems not working, and I'm not able to arm... any thought? Thanks

jflyper, 2017-02-15:
The BF-SoftSerial is ready and will be available from v3.1.6 patch release (1st, March).

GareFPV, 2017-02-17:
@jflyper Unify Pro v1 (Smartaudio v1) works great with 3.1.6 Build #55 on SoftSerial (SP3 Racing)

mhenrichsen, 2017-02-28:
@jflyper I see that the 3.1.6 build i out, but I don't see a option for the softserial on Unify pro. Can you give a pointer?

19Gerhard85, 2017-03-21:
@jflyper 
Any news on that? I would like to use SmartAudio on Softserial on my X-Racer F303 V3.

flounderscore, 2017-03-21:
Also see https://github.com/betaflight/betaflight/issues/2640

lasharor, 2017-04-09:
Can confirm Smartaudio v1 working with Softserial (SPracing3) after remapping the softserial tx port to motor output 7

t2000, 2017-06-16:
@lasharor: May I ask how your setup looks like? I want to attach my vtx to the softserial2 of my SPRACINGF3 and soldered the pin6 of I/O2 (the socket next to the beeper pins) to the microphone pin of the vtx.

My resources look like this:
```
# resource
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
```

I have configured softserial2 to be used with peripheral TBS smartaudio but once I do this telemetry stops working completely for me... I have to say that softserial1 rx AND tx are soldered to the smartport of my frsky fr4rsb, but that is working fine without the new vtx...

What am I doing wrong?

lasharor, 2017-06-16:
Try arming your quad...

t2000, 2017-06-16:
I just tried to arm the quad but it doesn't help. Also when softserial2 does not have tbs msrtaudio configured my taranis receives telemetry (like heading + pids) if the quad is in disarmed state.

If I configure the softserial2 to use tbs smartaudio telemetry stops working completely. Could it be a problem that I am connecting the smartport of my x4rsb to softserial1?

t2000, 2017-06-16:
Alright, I "fixed" it:

```
resource MOTOR 7 NONE
resource SERIAL_TX 12 A02
```
and then solder the audio cable onto the motor 7 pin. It really looks like that the I/O2 pins do not work...

Thanks again @lasharor

jflyper, 2017-06-16:
@t2000 The reason your original pin configuration didn't work is that all 4 pins you designated as software serial are different channels of the same timer. What it means is that two software serial ports must run at the same speed. SmartPort telemetry and SmartAudio use completely different speeds.

t2000, 2017-06-16:
i see that makes sense. Thank you for the clarification. But I still do not understand why this setup is the default in the SPRACINGF3 for the 2 softserial ports... Either there should be a warning that both ports have to operate on the same speeds or they should have configured one softserial onto motor 7 or 8 since most racers only have 4 motors anyway...

jflyper, 2017-06-17:
Well, it has been like that since NAZE!!! ;)

skorzo, 2017-07-10:
@t2000  @jflyper   I have a similar problem on my X-Racer F303 3.1

Uart 1 = USB
Uart 2 = Smartport telemetry (working ok)
Uart3  = SBus

And I need to make the Tbs smartaudio work.

According to the documentation the board has its softserial on:
RX = PWM 5
TX = PWM 6

And according to https://github.com/betaflight/betaflight/issues/1065 that softserial only works when a 10k ohm resistor is connected on softserial rx/tx line.

So my config went to:

Uart 1 = USB
Uart 2 = Smartport telemetry (working ok)
Uart3  = SBus
Softserial 1 = PWM5 - PWM6 - 10k resistor -> Smartaudio TBS

This config almost works.  Smartport and Smartaudio work but not in a consistent way,  I lose smarport or smartaudio most of times.

So I was adviced to remap Softserial TX from PWM6 to  Motor 7 pin because this config seems to work really OK.

But now...my problem is that Motor7 seems to "share" resources with  Serial_Tx 2

In my "dump all" I can see:
**resource MOTOR 7 A02 
resource SERIAL_TX 2 A02**

So I guess that Motor 7 pad is A02 ....isn't it?

If I try something like:

resource Motor 7 none -> Ok, Motor 7 freed
resource SERIAL_TX 11 A02 -> Error message and Serial_Tx 2 disabled  :-(

The weird thing is that after those changes Smarport is still working but TBS audio connected on Motor 7 softserial is still not working

Any clues please?

jflyper, 2017-07-10:
@skorzo 

BFSoftSerial no longer needs RX side wiring. Just connect TX wire and you are good to go.
With this in mind, try PWM5 **OR** PWM6 again.

skorzo, 2017-07-11:
@jflyper   Yesterday I tried PWM6  and it works OK with Smartaudio, or at least by now.

This afternoon I'm going to do more tests on the field, I hope that everything goes ok. Probably I was a little obssesed with the issue referring problems on softserial default pads and the need of a resistor, and I didn't give it a chance.

Thanks!

jflyper, 2017-07-11:
@skorzo 
You know, it's written in the BF wiki...

https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio

> Wiring
Just wire the SmartAudio wire to a free TX port, hardware UART or software serial. (There are some compatibility issues reported; if you have any problems, search the net before going into the BF github repo and cry for help.)

closed 2017-07-26 by jflyper