betaflight/betaflight #2179: Betaflight 3.1 rc10 on cc3d mwosd not working

dronejunkie, 2017-01-22:
Here the spec
betaflight 3.1 rc10 
FC is the falcon 250 cc3d first generation.
mwosd firmware is R1.6.6.3 

I could see mwosd loaded on the monitor but could not display any telemetry. The video are ok. With exactly the same setup it works after I flashed with betaflight 3.0.1. I flashed mwosd to R1.6 with betaflight 3.1 rc10 the same only this time it says no data on the monitor but the video is working. I tried the main and flexiport. Both are the same.

mikeller, 2017-01-22:
Can you check if MSP is enabled on UART1 under RC10? It used to be enabled by default, but that was changed since it had been causing problems for certain configurations. If it's not enabled, enable it and your OSD should be fine.

dronejunkie, 2017-01-22:
Yes, I have enabled MSP in uart1 and also telemetry.  Same configuration
work when I flashed back to 3.0.1.

J

dronejunkie, 2017-01-22:
The first uart I tried was 3 (mwosd chip connected to Plexiport) with MSP enabled and enabled telemetry in the configuration tab. I have serial enabled in uart 1 for sbus. Only main port has the sbus inverter. When that failed just trying different things I connected mwosd to main port , disabled serial and enabled MSP. No luck. So I flashed back to 3.0.1. it worked but 3.0.1 does not work  when I tried to use the plexiport, which is strange also. Should it work with plexiport?

dronejunkie, 2017-01-29:
Tried on 3.1.0 final still broken.

VvIRONWOLFvV, 2017-03-15:
bf v3.1.6 still having this issue with flexiport osd........none of the 3.1.x have worked for uart3 osd on cc3d

dgrigg123, 2017-03-16:
Same issue here.  Glad I found this thread.  I thought I was going crazy.  I tried putting minimOSD on Flexi and main.  No data.  I too am running SBus on Main, but tried to switch by putting SBus (Serial) on Flexi and try OSD on main.  Still no data.  I have not tried 3.0, but this sounds like an actual issue now.

VvIRONWOLFvV, 2017-03-18:
Yeah had to go back to librepilot until bf fixes this issue, hope Boris gets on this soon

dgrigg123, 2017-03-23:
I don't want to go back to libre just for this reason.  I was really wanting to redo my PID tunes over OSD in the field with BF.  That was one of the main points for me.  I'll wait a little while longer, but it would be nice to know if it is in the bug tracker list for Boris.

VvIRONWOLFvV, 2017-03-23:
I hope it is aswell, I've met a few with this issue, I even tried cleanflight and I get the reverse, osd info works great on flexi but it doesn't let me put the uart1 on serialrx and frsky at once which won't let my sbus receiver work

closed 2018-04-21 by DieHertz