betaflight/betaflight #2180: Betaflight 3.1 RC10 - SPRacingF3 w/TBS Unify HV v1.0

cbnlnk121, 2017-01-22:
Hi all, I just updated my SPRacingF3 with 3.1 RC10 and have a TBS Unify HV v1.0. The audo from the Unify is currently set to the TX of my UART3. I have also setup the Peripherals in Ports to the SmartAudio dropdown. I've tried to do all the ways to get the OSD to show up and I'm not getting anything.

Throttle center, pitch down yaw left and right. Nothing happens. 

Is there some other setting I need to use to get the OSD for smartaudio to work? Is it required I use a MWOSD or Minim?

jflyper, 2017-01-22:
@cbnlnk121 
I've responded you in the RCG thread.

cbnlnk121, 2017-01-22:
Issue solved .. I need to use my MWOSD .. misunderstanding by me. Thanks for the help. CLOSED

closed 2017-01-22 by cbnlnk121