betaflight/betaflight #2181: BF 3.1 build 10

digitalomni, 2017-01-22:
can not save filter settings through betaflight configurator gui or CLI.

mikeller, 2017-01-22:
Can't reproduce, filter settings save just fine for me. Make sure you've got the latest version of the configurator (1.9.0).

E120103016, 2017-01-22:
The latest version of the configurator (1.9.0). Has been improved # 2169.the configurator (1.8.9) problem. Today 2017.01.22 pm set to save everything ok! Very easy to use. Thank you!

closed 2017-01-22 by mikeller

digitalomni, 2017-01-22:
I have the latest version of the configurator 1.9

I am trying to change the Gyro Notch Filter Cutoff 1.   from 0 to 250

It will not save.

mikeller, 2017-01-22:
@digitalomni: What is your gyro notch 1 cutoff? If the cutoff is higher than the filter, the filter will be reset to 0 when saving...

digitalomni, 2017-01-23:
@mikeller My apologies.. Thank you