betaflight/betaflight #2184: IRC Tramp control through switches

noamsmadja, 2017-01-22:
could be cool to add control to the Tramp using Switches. Maybe use a knob on the remote to switch through a list of 4-6 favorite channels. Or 2, 3-pos, switches to change band and chanel kind of like the pid rates in the adjustments tab.

frizzle101101, 2017-01-22:
wait they haven't already implemented that? i thought the main purpose was to change ur vid channel on the fly from ur transmitter...

raphaelcoeffic, 2017-01-22:
You can control your VTX using OSD. Further control possibility will come in later releases. Please note that we delivered this feature only one week after we received some info on the protocol from IRC. And this was already late in the RC phase.

noamsmadja, 2017-01-22:
I was not complaining, is there already an open issue for further development for the tramp? Ill be happy to help contribute where possible

raphaelcoeffic, 2017-01-22:
> 
> I was not complaining, is there already an open issue for further development for the tramp? Ill be happy to help contribute where possible

No worry, this will be taken care of ;-)

andymac4182, 2017-01-23:
@frizzle101101 This is implemented for the Singularity I believe (It was in 3.0.1) so just needs to be adapted to the new interfaces.

jflyper, 2017-01-23:
@andymac4182 Yes. Stick and switch commands from singularity will be applicable to all VTXs.

closed 2017-01-31 by borisbstyle

Jama85, 2017-02-02:
This is now working? can i change tramp setting by switch or knob?

jflyper, 2017-02-02:
Not yet. I'm working on it (in parallel to single-wire software serial that is as demanded as the stick/switch UI).

TempleClause, 2017-02-07:
@jflyper Do I understand correctly that your solution will work with any radio?

jflyper, 2017-02-07:
@TempleClause 
Yes.
The sub-system (will probably be called VTXRC) is the generalization of the SINGULARITY mode. See the [SINGULARITY manual](http://singularity.impulserc.com/assets/files/manual.pdf) to learn how it will  (likely) to work.