betaflight/betaflight #2185: Betaflight 3.1.0 RC10 FPV-angle-mix is partially broken

nhadrian, 2017-01-22:
Hi,

Today I tested my quad-X 170mm racer with the latest RC, flight controller is SPRacingF3 (clone).
It looks like FPV angle mix is partially broken, the mix only works in YAW axis but not in ROLL axis.
I tried with multiple setup angles (roll_yaw_cam_mix_degrees) to double check my CLI settings (from 15 to 25 degrees). 

Regards
Adrian

jflyper, 2017-01-22:
I thought GIMBAL_TILT is to be mixed with ROLL and PITCH...

nhadrian, 2017-01-22:
FPV angle mix is not related to gimbal or servos!
It is a virtual coordinate system related to the FPV camera angle. 
Here is a description: https://www.youtube.com/watch?v=gXeaINFpvow&feature=youtu.be

jflyper, 2017-01-22:
@nhadrian Ah! Apologies!

borisbstyle, 2017-01-23:
Can you post a blackbox log please?

nhadrian, 2017-01-23:
Hi, I did a short test flight, please note that there is dark in Hungary now so I couldn't fly as smooth as I use to. But tried to add clean roll and yaw inputs to show the behaviour.
[BLACKBOX_LOG_NHAdrian_20170123_173500.zip](https://github.com/betaflight/betaflight/files/724324/BLACKBOX_LOG_NHAdrian_20170123_173500.zip)

borisbstyle, 2017-01-23:
Sorry forgot to mention. please enqble "set debug_mode = anglerate" and
make a log

nhadrian, 2017-01-23:
Done. 
[BLACKBOX_LOG_NHAdrian_20170123_191419.zip](https://github.com/betaflight/betaflight/files/724559/BLACKBOX_LOG_NHAdrian_20170123_191419.zip)

borisbstyle, 2017-01-24:
Issue resolved

closed 2017-01-24 by borisbstyle