betaflight/betaflight #2186: BFF3 OSD SW mode not working.

woody37, 2017-01-22:
Not sure if this is a feature request or a fix but would like to have this feature to instantly turn off OSD during flight for racing as well as DVR recording.

jflyper, 2017-01-22:
@woody37 
The OSDSW was designed for switching overlay layouts on external serial connected OSDs, and is not linked to the FC-integrated OSD atm (the FC-integrated OSD doesn't support multiple layouts anyway).

So you want two separate features,

1. Ability to select FC-integrated OSD overlay layout by OSDSW.

I don't know about this one

2. Ability to control (start/stop? on/off?) the onboard DVR equipment.

woody37, 2017-01-22:
The DVR was an example or purpose for the need of having ON/OFF OSD display during flight.  To clarify, I was requesting only **1 feature** and that was to turn on/off OSD by an assigned switch.  I like to fly with a clear screen and then be able to periodically manually switch on OSD display on to see data.

jflyper, 2017-01-22:
@woody37 Ack.

woody37, 2017-01-22:
Not sure of Ack, but the correction above is right on!  Thank you for responding.

borisbstyle, 2017-01-27:
@jflyper Any ideas if this can be restored easily?

DanNixon, 2017-01-27:
#2235 should do the trick.

borisbstyle, 2017-01-29:
Added to 3.1.1

closed 2017-01-29 by borisbstyle