betaflight/betaflight #2187: RC10 infinity rolls of death

npmullins, 2017-01-22:
getting infinity rolls of death with high rates using xracer_spi, betaflight RC10, 8k/8k, dshot600, racerbee v2 escs.
Doing multiple rolls/flips can bring it on, or hard snap rolls/flips
Lowering the rates stops it happens so often.

AndersHoglund, 2017-01-22:
Is it this ESC? : http://www.readytoflyquads.com/30a-superracerbee-pro-esc-blheli-oneshot
If so, did you order them with the option for Dshot?  Quote:
> Now with the Option to have D-shot firmware Flashed by default and the cap removed to work prefect the D-shot Protocol.

If not, have you modified and removed the signal capacitor on signal input?
If not modified, try Dshot150. And If that does not work, remove the capacitor.

npmullins, 2017-01-22:
i removed the signal caps myself, as it was an easy job, and their was this guide on youtube to follow
https://youtu.be/X8PGQ4nF5Ko he points out the cap location and i have removed mine

AndersHoglund, 2017-01-22:
OK, and you are absolutely sure it can handle DS600? Have you tried how it behaves with Dshot300?
High bitrates require good signal conditions, not only in the ESC but also FC and cabling must up to the task. You do have twisted signa/GND cables to all ESCs? Try 8k/4k with Dshot300.
You should also try raising the motor idle rpm, by increasing the "digital_idle_percent " value. Classic solution for de-sync problems.

npmullins, 2017-01-22:
ok thanks i will try these settings next. 

currently only running signal wire only to the esc from the fc

npmullins, 2017-01-22:
Are you sure this is a desync issue? as the quad is not recovering from the roll/flip and continues to roll/flip when the radio stick is moved back to the centre. Many thanks

AndersHoglund, 2017-01-22:
No, I do not know much or anything about your issue. I am only guessing since you give us very little input to work with. Only one clue at a time, but also only after asking.... 
[image]

Cheredanine, 2017-01-22:
Inclined to agree with Anderson his approach is the classic to this type of problem, but bb log would be very informative, there have been issues with gyros overloading but think this is more at the 32k gyro settings, either way doesn't seem like a bf issue

borisbstyle, 2017-01-23:
Closing this one as it is not a firmware issue

closed 2017-01-23 by borisbstyle