betaflight/betaflight #2189: RC10 RC Interpolation not on all channels

TobiFPV, 2017-01-22:
At RC10 the RC Interpolation does not work on all channels. With RC5 there was no problem. I noticed that the RC_Interpolation has been optimized with RC6. So this behavior was caused by this change or maybee it is intentional?

Setup:
Omnibus F4 AIO V2
X4R-SB Receiver

RC10 with Auto Interpolation (19ms)
[image]

RC5 with Auto Interpolation (19ms)
[image]

Through the missing interpolation on throttle i get some micro vibrations caused by the "stairs" in the motor rampup.

Motor spectrum without RC Interpolation on throttle:
[image]

DieHertz, 2017-01-22:
You can re-enable interpolation for both throttle and yaw via CLI at expense of higher CPU usage

borisbstyle, 2017-01-22:
@TobiFPV 
This is a configurable option as mentioned in release notes. You can enable or disable yaw and/or throttle interpolation in the cli if you want, but yaw has no D and doesnt really need inerpolation

closed 2017-01-22 by borisbstyle