betaflight/betaflight #2195: Unify Pro 5g8 version 1 Smartaudio does not work with Betaflight OSD

dengorman, 2017-01-23:
Tested this with different UART ports without success. I also have an HV version which works perfectly.

jflyper, 2017-01-23:
@dengorman Apologies for inconvenience. Version 1 speaking device was not available at the time of development. It's now arranged, so the support is coming. Hope Chinese new year doesn't delay shipments.

dengorman, 2017-01-23:
Sorry, what do you mean by speaking device?

SoulPictures, 2017-01-23:
The 5V Variant of the Unify is speaking SmartAudio Version 1, the HV and Race Version are using protocol Version 2 (SmartAudio V2)

dengorman, 2017-01-23:
I see. So does that new code is coming to enable v1 to work?

jflyper, 2017-01-23:
@SoulPictures
> The 5V Variant of the Unify is speaking SmartAudio Version 1, the HV and Race Version are using protocol Version 2 (SmartAudio V2)

Not quite. Non-HV version is shipped with V2 for at least few months now.

@dengorman 
> So does that mean new code is coming to enable v1 to work?

Yeap.

dengorman, 2017-01-23:
Excellent. Thanks!

closed 2017-01-24 by borisbstyle

dengorman, 2017-02-08:
Has the new code to allow Smartaudio v1 to work been implemented yet?

jflyper, 2017-02-08:
@dengorman Waiting for V1 device to arrive in mail.

dengorman, 2017-02-08:
Roger that

GareFPV, 2017-02-17:
@dengorman Smartaudio v1 works great with Betaflight 3.1.6 Build-Nr. 55 and Softserial. Wont work on hardware UART on my tests.

jflyper, 2017-02-22:
@here Yes. I've found that V1 is compatible with 3.1.6 BF-SoftSerial.

Got a V1 device, the signal look identical to V2, but UART is constantly receiving garbage. Asking TBS for changes in line drivers, pu-pd changes and etc.

dengorman, 2017-02-28:
I guess this hasn't been resolved in the latest 3.1.6 release yet?

Mrono, 2017-04-04:
I have a V1 vtx and even with soft serial I'm unable to get it to communicate, I can hear the clicks and such in my goggles.

jflyper, 2017-04-04:
@Mrono 
(1) What is your hardware configuration (FC)? Which signal pad are you using?
(2) Can you see current settings in the VTX (SA) menu?
(2) If not, how does STATX menu item under VTX (SA) tells you?

jflyper, 2017-04-04:
@dengorman 
No, V1 is not compatible with hardware UART.

Mrono, 2017-04-04:
@jflyper
BF f3
Softserial TX1
No
It shows "off" with the baud and packet send count changing, all other values are 0. I can change it to V1 or v2 from off but it doesn't do anything

jflyper, 2017-04-04:
@Mrono 
Since you are hearing click noise, the signal is coming out from the pad okay.
I'm suspecting a problem on your V1 device.
Will ask in the RCG thread for anyone using the same combination.

advancetom, 2017-05-24:
Hey jflyper i have a version 1 also and same problem using softserial with BF3 FC no semms offline :(

wolverin3, 2017-06-03:
I'm also hvg the same prob. Tbs Unify Pro with Smartaudio v1. Wired up audio wire to PPM/B07 and mapped to serial_tx11. It does not work.

jflyper, 2017-06-03:
Those who have problem with getting SmartAudio V1 device on software serial online, please provide more info; FC, BF version and CLI diff.

Just saying "I have the same problem" doesn't do anything.

wolverin3, 2017-06-03:
Fc: betaflight f3
Bf: 3.1.7 and your 3.1.7 with fix

My setup.. wired up the smartaudio wire to b07 / ppm pin on fc. Added 'resource serial_tx 11 b07' to cli and enable tbs smartAudio for softserial1 under ports tab. It doesnt work.

dengorman, 2017-12-26:
Sorry to drag this back up, but does Smart Audio now work with V1 devices or not?

jflyper, 2017-12-26:
It has been working for half a year now.

dengorman, 2017-12-27:
Noted, thanks jflyper.

dengorman, 2017-12-27:
Update : had to map a spare pwm pin to Serial Tx 11 on my Airbot Omnibus f3 Pro board. But using Softserial with my Unify Pro on Smartaudio v1 works! Hurray!!

jflyper, 2018-01-02:
Happy props 😃