betaflight/betaflight #2196: Servo mapping is causing ESCs to be unresponsive

ameeuw, 2017-01-23:
SP Racing F3 board:

I activated SERVO_TILT and remapped the servo output using resource:

```
resource MOTOR 5 NONE
resource SERVO 1 B08
save
```

After enabling it, the servo adapts fine, but the motors are not properly spinning up. I am using a Racerstar 4in1 20A ESC v1.

mikeller, 2017-01-23:
This will cause a timer conflict since timer 4 is used for the motor 5 pad and also for the motor 2 and 3 pads. You should be able to get it to work by remapping motors 1 - 4 to a11, a12, b08, b09, and servo 1 to a06.

dragonclips, 2017-07-17:
Where i can know all timers and mapping resources?

jflyper, 2017-07-17:
https://github.com/betaflight/betaflight/blob/master/src/main/target/SPRACINGF3/target.c

dragonclips, 2017-07-18:
> main/target/SPRACINGF3/target.c

thank you

ameeuw, 2018-05-09:
The issue is solved. Thank you.

closed 2018-05-09 by ameeuw