betaflight/betaflight #2201: Mid-throttle oscillations at modest PIDs

CoryJRC, 2017-01-24:
Hello,

I am seeking some advice on tuning and the steps I should take to rectify an issue I am having on two of my builds.
I have conducted my own reading, of course, but have still not had any improvement that I can notice.

The builds:
*Armattan F1-5Acro with Armattan integrated FC (Naze 5, essentially). Aikon 30A V2, ZMX 2206 2300kv, Multishot.
* Rip off Alien. F3FC. Aikon 30A V2 (Cap removed). Racerstar 2205 2300kv. DSHOT600.

The issue:
The seemingly-classic mid throttle oscillation.
I have narrowed this oscillation down to almost only the yaw axis, I think (Blackbox learner). The oscillation seems to only occur at mid throttle (and neutral yaw?? maybe). The crafts are both entirely flyable, and generally respond very well to stick inputs. My PIDs are nothing out of the ordinary, as far as I understand. The shaking, though, is irritating, and I know I am not alone in experiencing this.

I have tried filtering the gyro, both in general (lowpass=70, then 50) and specific to yaw (yaw_lowpass=70, then 50) to no avail. I am not opposed to soft mounting motors, but am intrigued if this is the only option. I don't know enough to support or discredit the theory.
Attached is the blackbox log for the second craft, identified above, with no filters. I very much appreciate any input you may have, and I appreciate that you are busy people with your own problems to solve.

[nolfilter_pid1.bfl.zip](https://github.com/betaflight/betaflight/files/726519/nolfilter_pid1.bfl.zip)
[image]
[image]
[image]
[CLI DUMP – KNOCK OFF ALIEN.docx](https://github.com/betaflight/betaflight/files/726522/CLI.DUMP.KNOCK.OFF.ALIEN.docx)

Thank you.

borisbstyle, 2017-01-24:
Probably one of your motors causing this. Yaw is most sensitive to vibrations usually. From seeing these motors they indeed tend to easily cause that due to stronger magnets.
Softmount your FC and it will be better.

closed 2017-01-24 by borisbstyle

CoryJRC, 2017-01-24:
Thank you, Boris.  Did you see my blackbox log?  Also, is it possible that dialling the FC back BF2.6 would make this issue go away?  The reason I ask is that I have a build on an old version of BF, with motors (mismatched, in fact) that are dinged and far noisier than the brand new ones, and yet the old build flies with no issue like this.

I also tried using the 2.6.1 defaults that you have suggested, and this made no difference that I could see or feel.  Why is this so? 
2.6.1 defaults:
gyro_lowpass = 80
dterm_lowpass = 70
gyro_lpf = OFF

Thank you, Boris.

Cory

CoryJRC, 2017-01-24:
Also, is there any sense in using a notch filter in this example?

Thanks Boris.

borisbstyle, 2017-01-24:
Do you have blackbox logs?

CoryJRC, 2017-01-24:
Boris,

Yes. Please see the nolfilter_pid1.bfl that I attached to my original post.

Thank you.