betaflight/betaflight #2203: Blackbox SDCard Not Working?

HaseebNain, 2017-01-24:
Hey guys,

What is the status of Blackbox SDCARD recordings, I know there was the possibility of dropping blackbox serial implementation as it was considered "legacy" and detrimental to loop times, is this still the case with the SDCARD implementation? 

I am attempting to include the SDCARD drivers for blackbox recordings into a target (PODIUM), the pins have been defined correctly as well as the inclusion of the SDCARD feature into the target makefile. Is there anything else that needs to be included on the target side to enable blackbox recordings? I have looked through all of blackbox_io.c and blackbox.c and am not sure what else remains to be updated. 

This is the snippet of target.h that relates to the SDCARD 
https://github.com/ThrustUAV/betaflight/blob/master/src/main/target/PODIUM/target.h#L77 

    // SD CARD
    #define USE_SDCARD

    #define ENABLE_BLACKBOX_LOGGING_ON_SDCARD_BY_DEFAULT

    #define SDCARD_DETECT_INVERTED

    #define SDCARD_SPI_INSTANCE                 SPI3
    #define SDCARD_SPI_CS_PIN                   PB3

    // SPI2 is on the APB1 bus whose clock runs at 42MHz. Divide to under 400kHz for init:
    #define SDCARD_SPI_INITIALIZATION_CLOCK_DIVIDER 128 // 328kHz
    // Divide to under 25MHz for normal operation:
    #define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     2 // 21MHz

    #define SDCARD_DMA_CHANNEL_TX               DMA1_Stream5
    #define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG DMA_FLAG_TCIF5
    #define SDCARD_DMA_CLK                      RCC_AHB1Periph_DMA1
    #define SDCARD_DMA_CHANNEL                  DMA_Channel_0

    // SPI FLASH Pin Definitions
    #define USE_SPI_DEVICE_3 
    #define SPI3_NSS_PIN            PB3
    #define SPI3_SCK_PIN            PC10
    #define SPI3_MISO_PIN           PC11
    #define SPI3_MOSI_PIN           PC12

Some questions I have:

- There is no detect pin, however according to blackbox.c writing should begin as soon as the FC is powered if a detect pin is not assigned. Is this not how the blackbox writes? Should a specific detect pin be included?
- SD cards have a maximum write speed of 25MHz so having the APB1 clock speed divided by 4 brings it safely under the 25MHz limit at ~21MHz. There are a few targets (SIRENFPV and BLUEJAYF4) that are incredibly below those thresholds, is the timer I have set still set to fast?

The SDCard is never initialized according to the cli and comes up as a fatal error on the configurator's "Blackbox tab". 

[image]

Blackbox isn't my forte so if there is anything I should look into in order to get a better idea of this implementation, any sort of guidance would be greatly appreciated.

goatspoon, 2017-01-24:
Have you enabled it in "MODES?

HaseebNain, 2017-01-24:
Yes, I had to double check. 

It is set to be enabled at the same time the quad is armed.

borisbstyle, 2017-01-26:
Looks like sd card is not detected.

borisbstyle, 2017-01-26:
You will have to use hardware debugger to find out where it fails.

MennoBr, 2017-01-30:
Have you tried a lower logging rate ? 100% on 8K is pretty high

HaseebNain, 2017-02-02:
@borisbstyle The data to the SD Card and the clock is present on the scopes, there is no data from the SD Card. 

@MennoBr I have tried reducing the logging speed but to no avail.

I have checked some other targets and it does not seem a detection pin is mandatory as other targets seem to have SD implementation without the use of that pin.

I will dig deeper into the blackbox classes and more into how to implement the SD card into targets, any references or documentation I should look into?

unitware, 2017-03-24:
I see a similar problem on my spracingf3neo, card is not accepted. Was fine with 3.1.6
I have tried to erase all files on card, and also a quick format using the sd-card formatter tool.

mikeller, 2017-03-24:
@unitware: Have you checked to make sure that the feature SDCARD is enabled in your config? This used to be hidden and enabled by default.

unitware, 2017-03-25:
@mikeller thanks, I have added that as a default feature for that board now. (https://github.com/betaflight/betaflight/pull/2740)

closed 2018-04-21 by DieHertz