betaflight/betaflight #2206: Support for Connex Pro sight HD

frizzle101101, 2017-01-24:
I'm not sure what's already done on this topic but I have obtained a conned pro sight HD fpv system and I was reading the manual and it says to connect to a UART on a flight controller and use it as mavlink. I'm not sure what can be done with this currently but I was hoping there was some ppl out there egar to improve the workings of these 2 systems to improve the osd features on the connex pro sight with betaflight information.... Thats

frizzle101101, 2017-01-24:
Sorry to finish. Thats just an idea to put out there hoping people as hyped about HD fpv as I am

andymac4182, 2017-01-24:
@frizzle101101 Have you tried using the mavlink telemetry in betaflight?

digitalentity, 2017-01-25:
@frizzle101101 ProSight OSD is quite limited. Even with telemetry it will only show:

 1. Flight mode (you probably don't need OSD to know that you are flying acro don't you)
 1.  Heading (useless if you don't have a working compass)
 1. Battery voltage (it does show voltage even without telemetry).

No current, no GPS coordinates, no speed, no altitude, nothing that is of real use. AFAIK there are plans to improve that, but I have no idea when it will happen.

closed 2017-01-26 by borisbstyle