betaflight/betaflight #2215: Configurator Mixer Settings

blckmn, 2017-01-25:
One for 3.2

Allow target maintainers to add default mixer settings (resources etc) to be added to configurator for a given target. This would then prompt the user to "setup" those settings whenever they change the mixer (the user can accept or reject).

@AndersHoglund this should solve the messy users need to know the ins and outs of a board to use resource - but also puts this back on target maintainers.

It will require resource mapping via configurator (MSP or alternative).

mikeller, 2017-01-25:
Totally agreed that this is the way we should be heading.

My thoughts so far:
- use CLI based backup / restore for the configs, at least until we get a way to do complete binary config dumps through PG (even then I like the fact that CLI dumps are human readable);
- do it indirectly, i.e. all the target maintainer has to supply is the URL for a config file. The config file contains a list of versions, each one specifying a list of configs (like 'tricopter', 'quad beginner', 'quad racing',...), each one of them with a link to the actual config file.

Thoughts?

AndersHoglund, 2017-01-25:
Jepp. This is one way fwd. But there are a few questions, and doubts.  The responsibility are push on the Target Maintainers, but who are "they"? In many cases it is still "we". We still can't just drop responsibility and just assume some one else will pick it up. "SEP" (Somebody Elses Problem) is no good attitude.
 
Dynamic resource management.  When configurations change on target, requiring resources to be re-allocated, are there or will there be functions to do that? For example the "SERVO_TILT". Previously that shifted motors two steps for resource conflict reasons. Automatically. Will that work in a similar way? There are more cases like this, just one example. SerialRX, PPM, PWM -input re-config is another.
Or will we have a number of pre-configured settings for different scenarios, much like the RF boys now have preconfigured binaries (_STD,  _SPEKTRUM_TELEMETRY). I.e. REVO_SERVO_TILT.cfg, REVO_PWM.cfg and even REVO_PWM_SERVO_TILT.cfg. The number of permutations will be endless..... There must be a more systematic way.

closed 2023-02-24 by blckmn