betaflight/betaflight #2217: 3.1 Naze - 3 motors running full speed

ShikOfTheRa, 2017-01-25:
3.0.1 + cleanflight run OK

with 3.1, three motors run flat out as soon as arm.

Simple config:
Naze target, wiped eeprom settings
calibrate acc
set rx to ppm and enable serial2 and that's about it.

moved back to 3.0.1 and ok
put 3.1 on again and wiped. 3 motors flat out on arm.

Not touched any of loop settings etc.

Any ideas?

blckmn, 2017-01-25:
Can you please post a CLI diff?

ShikOfTheRa, 2017-01-25:
ok will do tomorrow. late in day here... and i've reverted back down

johnty, 2017-01-26:
i'm experiencing similar behaviour on a naze32. actually one motor goes flat out on arm as soon as the minimum threshold is exceeded, other two jumps to some intermediate value. the third looks "almost" normal (although this is my first build so... ;) ). it seems like stick input and accelerometer (if switched to angle mode) will still have some effect, but the primary value is stuck on these constants.

reverted to previous version and problem goes away.

this is testing without ESCs, btw. (i did try one ESC on PWM mode for testing, but it seems to be independent of whether they're connected or not).

also i find that after a clean erase/flash of firmware, it would be working normally for a little while, but after a bit it gets stuck in this mode it seems.

i can post some more info when i get home later if needed.

ShikOfTheRa, 2017-01-26:
reflash with wipe.
defaulted settings and following is diff. 

# diff
# Betaflight / NAZE 3.1.0 Jan 25 2017 / 09:42:22 (88693ca)

set motor_pwm_protocol = OFF
set pid_process_denom = 2
profile 0

rateprofile 0
rateprofile 0

#

borisbstyle, 2017-01-27:
Can you try oneshot125 for example?

ShikOfTheRa, 2017-01-27:
Unfortunately I don't have any ESC that support it yet. Yeah I know -  living in the dark ages !! 
I will be updating at some point soon.

ShikOfTheRa, 2017-01-27:
Further testfeedbak - may be relevant:ck
Using Motors tab, the motors work as expected. Gradually increase in speed in line with motor sliders

Motors 3+4 appear flat out - Burned motor out in testing :(
Motor 1 appears to be influenced by gyro, but running maybe 50%
Motor 2 not running

Looking at RC inputs - all seem OK. No strange values. The quad model in RC tab in gui rotates exactly as expected from rc inputs.

The main model in setup tab appears correct and moves correctly as rotate in all axis.

mikeller, 2017-01-28:
@borisbstyle: OneShot125 works fine on NAZE rev6, I am using that on my tricopter.

Noctaro, 2017-01-28:
Hi, i am having exact the same problem with Naze. I first tested it without props, using standart looptimes (1k/1k), seemed to work. I then set looptime to 1k/0.5k and forgot to test arm it ^^´ 
With props on it went upside down instant. 3 motors flat out and 1 does not move (its the right rear motor that does not work.  (BF 3.1 stable)

# diff
# Betaflight / NAZE 3.1.0 Jan 25 2017 / 09:42:22 (88693ca)

feature -RX_PPM
feature RX_PARALLEL_PWM
feature BLACKBOX
set min_throttle = 1090
set baro_hardware = AUTO
profile 0

set p_pitch = 80
set p_roll = 60
rateprofile 0
rateprofile 0

set rc_expo = 15
set rc_yaw_expo = 10

#

johnty, 2017-01-28:
here's what it looks like in the most version of [cleanflight](https://www.youtube.com/watch?v=4B79ydk7DE8)

compared with [betaflight](https://www.youtube.com/watch?v=uQPfBC8lDXs), where the motor output values get stuck as soon as it is armed. i actually experience this in versions 3.0.1 and 2.91 as well, not just the latest!

settings after a clean flash:

- serial rx on UART2, ibus receiver
- aux switch for arm
- aux switch for acro/angle mode

CJ-Davies, 2017-02-02:
Same issue here on a rev6 using 3.1.3. Motors respond normally when moving the sliders in the motors tab, but when arming they go to;

1 - 1515
2 - 1030 (this is the only one that behaves correctly)
3 - 2000
4 - 1515

nitr0123, 2017-02-03:
Happening to me too. Naze32 running ppm via d4r-ii. Had been running this on something like 2.8. After updating motor 3 goes full throttle and others are half throttle. My observations mirror CJ-Davies's.
Fwiw I believe my naze is a genuine 6dof one.

Indy6, 2017-02-05:
Hi,
I honestly didn't understand the original post, I react to CJ-Davies comment, that is looking a bit like mine...
Since I flashed Betaflight 3.1.3 (before I had 3.0.1), motors respond accordingly when moving the sliders in the motors tab, but when I push the throttle, the motors are stuck to the min throttle value...???

ragd011, 2017-02-05:
I confirmed this with a CC3D.  The .8/.8 made a major difference.  With 3.1.3 flashed at default speeds when I armed and then throttled up (motor spin off) motor 3 was 100%, 1+4 were 50% 2 was 0.  After setting to .8/.8 everything responds better.  I'm only on the ground now with props off.  

Should I start another issue CC3D specific or are the FCs similar enough that here makes sense?  Thanks Rob.  BTW, I have BBL of both ways running 33%.

AussieInAZ, 2017-02-05:
I am using the Airbot Flip32 Revo 4 board and am having a very similar problem on BF 3.1. with an X4R using S.Bus.
I have tried 125, 42, Multishot and PWM.  The PWM mode won't even initialize the ESCs but the other modes get the same, or similar, results with one motor at 100%, two at mid throttle'ish and one at next to nothing.  This happens only when the ARM switch is flicked on my Taranis.  The motors work fine with the slider in BF.
I'm very interested in a fix if anyone comes up with something.

Cheers
Aussie In AZ

[image]

[image]

jflyper, 2017-02-05:
@blckmn @borisbstyle @ShikOfTheRa @AussieInAZ and everybody else:

I believe it's related to #2322.

AussieInAZ, 2017-02-05:
For anyone else with the Airbot F4 Revo 4 board from Ready To Fly Quads (or the like), I did a re-flash using the AirbotF4_3.1.3 BF firmware (it had Revo4 FW 3.1 before) and set up a brand new model in my Taranis.
All seems to be doing fine now on the bench and also by it's self (without PC connected) and the AUX arm switch activated.
I have no idea what my problem was but it's all working as it should (seemingly) now.
Now for my maiden test flight 

Aussie

amelluk, 2017-02-05:
@AussieInAz what esc protocol are you using where it works fine? Non-PWM? Can you confirm if the problem still exists with PWM escs?

blckmn, 2017-02-05:
This may be fixed in https://github.com/betaflight/betaflight/pull/2306 anyone willing to test the build?

amelluk, 2017-02-05:
Tested on master + #2306  - works great on the bench and fixes my issue #2322 for OMNIBUSF4SD , not flight tested yet but this fix puts me in the air at least.

AussieInAZ, 2017-02-06:
@amelluk  I am using these ESCs, "Aikon SEFM 20A BLHeli_S ESC (2-4s )" ( http://www.getfpv.com/aikon-sefm-20a-blheli-s-esc-2-4s.html ) and the motor protocol of OneShot125 in Betaflight.
For some reason, PWM will not initialize these ESCs (no final beep whan plugging in the battery) so I can't tell you if the motor problem still exists on PWM

Aussie

mikeller, 2017-02-07:
@AussieInAZ: Have you enabled PWM in the blheli config? PWM is disabled by default for later versions of blheli.

AussieInAZ, 2017-02-07:
@mikeller  I don't think I'm too concerned with PWM at the moment.........unless you think I should?  Oneshot125 seems to be working just fine (I think).  My main problem was the motors doing strange things when arming from my Taranis which it isn't now.

mikeller, 2017-02-07:
No I don't think PWM has a lot of relevance in your use case. My comment was just a possible explanation for why PWM does not initialise the ESCs.

ShikOfTheRa, 2017-02-07:
@blckmn 
Tried compiling your pwm_rate_check branch. Is this the correct one to try? Seem to get same issue

johnty, 2017-02-07:
@blckmn @ShikOfTheRa 

ditto. i checked out the pwm_rate_check branch from @blckmn and did a compile for the NAZE target. flashed it (naze10dof) and had exact same behaviour.

what I noticed however is that using the current latest stable version, using 0.8k/0.8k rates seem to be OK as others have suggested elsewhere.

amelluk, 2017-02-07:
We might be talking about two different problems in this issue? The pwm fix #2306 is for motors going full throttle. @johnty @ShikOfTheRa is this what you are experiencing? When I did a build with #2306 it fixed my problem.

johnty, 2017-02-07:
@amelluk: yes, perhaps they're different, as the motors are not _all_ going full throttle, but getting stuck around certain fixed points (one of them tends to go full throttle...)

ShikOfTheRa, 2017-02-08:
@amelluk 
When I raised this I thought three motors were full throttle and one not working. in my defence....props still attached on mine so hard to tell with all the noise! 
I posted a bit more detail further up, but it seems like it is like others reported... Some full, some influenced by gyro and one not spinning.

Feng94, 2017-03-26:
I believe I have the same problem on Omnibus F4 3.1.7 Betaflight. 
On Arm with Airmode this happens:
[image]

On 3.1.X Where X is a lower number than 7, this didn't happen before, had about 10 flights without issue, then suddenly it does this all the time without touching the hex or flashing anything. I did change the settings for the Notch Filters though.

Tried reflashing to 3.1.7, issue persists.
Tried different motor_pwm_protocols, issue persists.

Here is my dump:
http://pastebin.com/aZB1ezfB

Anyone have any luck?

ShikOfTheRa, 2017-03-27:
I reverted to earlier release 3.0 
Patiently waiting until this has a resolution, but not sure anyone is actively working on it

CJ-Davies, 2017-03-27:
I'm keeping my Naze copters on 3.1.3 as it's the last 'stable' version I've
found.

Feng94, 2017-03-27:
For me 3.1.0 is the latest 'stable' version. Thanks, managed to get in the air again, hopefully this is looked at for future revisions.

ShikOfTheRa, 2017-04-20:
I can confirm using latest release on NAZE works for me.  

Suggest to close?? I will if no feedback in next few days

grumezastefan, 2017-04-24:
The same problem for me...i have a zmr250 with naze32 recently i updated the firmware on betaflight because not have acces to use the program (old firmware)
When i arm by sticks motors starts to spin crazy :(

jflyper, 2017-04-24:
@grumezastefan We need to know the firmware version.

ShikOfTheRa, 2017-04-24:
@grumezastefan 

For me - and this issue:
3.1.0 not work
3.1.5 OK
3.1.6 OK
3.1.7 OK (latest at this time)

I am now using 3.1.5 and it works well. For .7 I had issues with #2852

grumezastefan, 2017-04-24:
I have Betaflight Naze 3.1.7 April 3 2017

grumezastefan, 2017-04-24:
I will try now 3.1.5 to se how its work.

grumezastefan, 2017-04-24:
Now i get the message no response from the bootloader...how to fix this?