betaflight/betaflight #2220: Can't flash with full chip erase on Omnibus F4

kf4mot, 2017-01-25:
Title pretty much says it.
With full erase off it's fine. With it on it hangs after the erase bar finishes and as the flashing progress bar comes up.

mikeller, 2017-01-25:
Duplicate of https://github.com/betaflight/betaflight-configurator/issues/341 .

closed 2017-01-25 by mikeller

AndersHoglund, 2017-01-26:
Well, 341 is claimed to be fixed by 344. Not quite so....This is still a problem. Or there may have been a few different problems, where one was fixed. This is more a duplicate of https://github.com/betaflight/betaflight-configurator/issues/200

TwoToneEddy, 2017-07-11:
This is definitely still a problem, I have the same issue and always have had.

sqweezr, 2017-07-29:
I have the same problem, but if i now flash without a full chip erase, it reboots and wont start. I can only see it in DFU mode. How can i fix this?

TyTechSci, 2017-08-03:
Same issue here. Can connect fine to configurator, flashing says "Failed to open serial port"

imrj, 2017-09-26:
am having same issue here with F7....has therte been any fix for this??

Wes-Leigh, 2017-09-27:
Hi,

I had the same issue with Omnibus F4. Trying forcing Serial Speed and unplug RX if possible. That was the only way I could get mine to flash to RC5.