betaflight/betaflight #2222: BluejayF4 Blackbox death switch

EFAW, 2017-01-26:
Using the latest BluejayF4 build, turning on blackbox makes quad flip uncontrollably when armed and throttle up. Disabling Blackbox and removing Blackbox mode resolves problem.

borisbstyle, 2017-01-26:
I dont have this issue on mine. Just flew it this week.

Which build did work for you?

EFAW, 2017-01-26:
I am using the dhfcpdb board im assuming this is the problem

mikeller, 2017-01-26:
Where are you logging to? If it's to SDCard, try switching logging to onboard flash. There have been issues with BLUEJAYF4 logging to SDCard.

EFAW, 2017-01-26:
Logging to the onboard flash

EFAW, 2017-02-02:
3.1.2 and last beta still have same problem when logging to spiflash. Can anyone confirm whether it is an issue with all bluejay boards or just the boards based on BJF4?

HenryIvarsson, 2017-02-11:
Same here, logging to flash on BJF4 rev 3will make it into an instant ceiling rocket on arm. Logging to sd works great.

blckmn, 2017-02-21:
When you get the ceiling rocket, do you have an sdcard in - even though using BB onboard?

HenryIvarsson, 2017-02-21:
@blckmn No, I tried it last time when I didn't have an SD card handy. I can try both ways tonight on the current release and post a dump if that helps.

blckmn, 2017-02-21:
If you could that'd be great. Might help narrow down the issue.

HenryIvarsson, 2017-02-22:
@blckmn 

Hey guys, some findings. See attached dump with only change being setting BB to flash storage. crazy jitter. Please find a bb log of this attached. Then I change to SD and all good. I do this on 2 birds, same result. 

This is crazy though, per your question I set to flash but put a sd in slot (not a resonable user case), then it seems OK, less or no jitters? Puzzled I remove the SD and back to jitters. Se attached clips of all scenarios. My iphone got some crazy jitters also, apologies for that, I might have held it in an angle that it jumps between portrait and horizontal, no idea, never mind that.

Sorry for being a bit of a sissy and testing without props but I had already had both of them hitting the ceiling because of this lol.

https://www.dropbox.com/sh/eopoo3i1oxpb9ik/AABTaYqAWzIcrRbRu2zgwcXWa?dl=0

blckmn, 2017-02-22:
Hi there @henryivarsson - perfect thank you. I suspect there is still some cross over with the SDCARD or similar and it is causing issues when one is not inserted. Will try and resolve this bug this weekend.

HenryIvarsson, 2017-02-22:
Cool, let me know if there is anything else I can assist with. My coding skills top out at 10 Print "Hello World" 20 Goto 10 but have worked 10 years in IT with (technical) sales so happy to test. :)

blckmn, 2017-02-23:
If you disable the SDCARD feature (in CLI use the command `feature -SDCARD`) then do you still get the jitters?

blckmn, 2017-02-26:
This should now be resolved with recently merged PR please reopen if not.

closed 2017-02-26 by blckmn

RipperDrone, 2017-03-27:
@blckmn: Trying to get a blackbox recording set up - neither onboard nor SDCARD works with default settings BF3.1.6, always comes up with an error:

[image]

mikeller, 2017-03-27:
Have you tried reformatting the SD Card with the SD Card formatter (https://www.sdcard.org/downloads/formatter_4/) with 'format size adjustment' on?

Re the "Your flight controller does not have a compatible data flash chip available" error. This is a bug of the current firmware, causing the onboard flash only to be reported correctly if 'On-Board Flash' is selected as the logging destination.

RipperDrone, 2017-03-28:
@mikeller Thanks, your hint did the magic. Recording ok now :-)

mikeller, 2017-03-29:
@RipperDrone: Unfortunately it is a rather obscure fact that SD Cards should always be formatted in this way if they are to be used for streaming data (i.e. in most cases), and Windows and other operating systems will happily reformat them in a rather unsuitable way at the click of the mouse...