betaflight/betaflight #2223: CLI diff all command does not print mmix configuration if mixer=CUSTOM

mlopyrev, 2017-01-26:
Steps to reproduce:

```
# reset configuration to default settings
defaults

# mixer
mixer CUSTOM
mmix reset

mmix 0  1.000 -1.000  0.800 -1.000
mmix 1  1.000 -1.000 -0.800  1.000
mmix 2  1.000  1.000  0.800  1.000
mmix 3  1.000  1.000 -0.800 -1.000

save

# print diffs
diff all

# mixer
mixer CUSTOM
mmix reset

# servo
# servo mix
...
```

Note mmix configuration missing in the diff all output.

This took me a while to spot.
Symptoms: ESCs don't initialize, FC starts intermittently spitting out gibberish characters in CLI if mmix values aren't defined for mixer=CUSTOM (possible memory corruption?).

borisbstyle, 2017-01-26:
So the diff doesnt show custom mix value at all and it fails on printing the rest?
Can you make a screen/copy of that?

Why do you use custom mixer actually?

mlopyrev, 2017-01-26:
Hi Boris. Sorry it wasn't clear from my example. Cli diff prints the rest of the diff correctly. 

One the thing worth noting beyond just fixing cli diff to handle non-existant defaults is that the system had basically undefined behavior with mixer=CUSTOM and no mmix values specified. (the result of me pasting the output of 'diff all' to initialize the board after a flash/erase - and not noticing the mmix definitions being absent). 

Perhaps some sensible defaults on 'mmix reset' would help. Even better - show a warning message on 'save'. 

Further context:
  - frame: Speed Addict 250-R
  - board/target: X-RacerSPI

Thanks for looking into it!
I haven't coded in this project, but happy to give
It a crack if you think warning on save is useful.

borisbstyle, 2017-01-26:
The issue is already fixed in 3.1.1 patch (unreleased yet but scheduled in a week or 2)

mikeller, 2017-01-26:
@mlopyrev: The whole point of `mixer=CUSTOM` is that you have to add a user defined mixer config - as the manual (https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md) states:

>  Each motor must be defined with a custom mixing table for that motor.

If you want presets, then simply don't use a custom mixer preset.

mlopyrev, 2017-01-26:
@mikeller: of course, you are absolutely right. The API is well documented and clearly states that the mixing table should be defined.

However - and I mean absolutely no disrespect at all - good software engineering practices call for systems and APIs (and especially user interfaces) to be designed in such a way that they are easy to use and **difficult to misuse**.

In this particular case, an unrelated bug in defaults caused accidental (and unnoticed) misuse of the mmix API, resulting in one of your fellow FPV pilots spending a bunch of time debugging instead of flying. ;) Even after the defaults / cli diff is fixed, someone else down the line might fall into the same trap.

I'm hoping to leave this thread at constructive level: how do we make this awesome software even more awesome.

mikeller, 2017-01-26:
@mlopyrev:

> However - and I mean absolutely no disrespect at all - good software engineering practices call for systems and APIs (and especially user interfaces) to be designed in such a way that they are easy to use and difficult to misuse.

I totally agree with you on this. Unfortunately, when it comes to Betaflight, there is a trade-off between good API / UI design, and the space available on the MCU (and to some extent also time available for development). As it stands, we already have to limit the number of features available on all but F4 MCUs, and adding more comprehensive config validation would make this situation even worse.

What I could see as a reasonable and reasonable small improvement of the current situation that is in line with how validation is done for other parts of the configuration would be that, when the config is saved, and if a custom mixer is selected but no mmix settings entered, one 'pro forma' mmix setting is added to the config before saving. While not leading to a flyable config (nothing can really since a custom mixer means we don't know anything about the model), it would stop the undefined behaviour.

If you want, you can always submit a pull request for this change, outside contributions are much appreciated in Betaflight. ;-)

mlopyrev, 2017-01-27:
@mikeller - thanks! Point taken. Some validation can happen in the configurator tho, right? Or does it create problems with backwards compatibility?

What I've seen Betaflight configurator do in similar cases (i.e., invalid set of parameters provided in cli or in UI) is to revert right back to a safe default. It's not always clear that it happened, and the UI could show some error message. 

By that logic, mixer could just revert back to the default preset if mmix is underspecified or invalid. 

Let me known what the preferred solution is and I'll give it a crack.

mikeller, 2017-01-27:
@mlopyrev: Validation does happen in the configurator where possible. A lot of settings are dependent on the board type _and_ on the firmware version, and there is not (yet) an API to query the board for capabilities (and we've avoided until now to keep a matrix of all versions by board in the configurator). So settings that are board specific tend to fall back on the validation / reverting to defaults that is done in firmware.

As for the custom mixer settings, doing validation in the configurator is not possible, since there is no API for the configurator to query or modify mmix settings.

Yes, I think your suggestion to revert the mixer to default if a custom mixer is selected and no mmix defined is better than mine to add a 'dummy' mmix. Thanks for offering to help.

closed 2021-02-17 by github-actions[bot]