betaflight/betaflight #2225: Indicator leds dont work right

moscher, 2017-01-26:
After some testing with the OMNIBUSF4SD target (Bf 3.2.0-989 and 1.9.0 configurator)
i get the same behaviour like descriped in #1271.

When i roll right, all! indicators (north east south and west) blink slow than faster.
When rolling left nothing flashes. While rolling left + pitch down every indicator flashes too. Same as rolling right.

mikeller, 2017-01-26:
Can you please upload a screenshot of the LED configuration tab for your config?

moscher, 2017-01-27:
This is my led-modul: http://www.helipal.com/bluesky-rgb-led-board-ws2812b.html

[image]

mikeller, 2017-01-28:
Ok. This only shows indicators for left and right, and they should work in this configuration. In general, can you try moving the LEDs to the middle of the grid? Symmetry *does* matter for the LED configuration.

moscher, 2017-01-28:
In the middle position its the same thing. Maybe i will get some other ledstripes from a friend to test them.

mikeller, 2017-01-29:
Can you again describe what works and what doesn't please? The configuration that you posted a screenshot for has only indicators configured for left and right, and they do look to be set up correctly.

moscher, 2017-01-30:
I tested with other LEDs but the same error.

I use an XSR via S.BUS, Taranis Mode 2.
My board is build in with the arrow backwards. So i changed the orientation to Gyro 90° CW, Accel 90° CW. (im wondering why not 180°?)
 In the receiverTab and preview picture all is everything is fine!
 Sticks are calibrated to 1000-1500-2000. I selected TAER1234.

When i move the
right stick left -> nothing blinks
right stick right -> both sides blinks
right stick up -> nothing blinks
right stick down -> both sides blinks

moving left Stick-> nothing blinks

i tryed without changing the board orientation, but i have the same error.

mikeller, 2017-01-30:
The board orientation for gyro and accel only affects how the signals for these sensors are interpreted, nothing else, and certainly not how the LED indicators work.

Before you try to set up the indicators, first make sure that your build is configured correctly, i.e.
- the model on the 'Setup' tab moves in the correct way when you lift up and rotate your model around all axes;
- the model on the receiver tab moves moves in the correct direction when your RX and TX are powered up and you move the sticks.

Only when all of this is set up correctly should you be looking into getting the LEDs to work correctly.

moscher, 2017-01-31:
In the setup and receiver tab everything looks good and behave like it is wanted.

It seems like im alone with the problem, thats weird.

My FC is an Onmibus F4 Pro.

Eganwp, 2017-01-31:
I assume you are plugged into PWM6 and not the old LED pads on the Omnibus F4 Pro, correct? This was changed with 3.1 so the silkscreening on the board for LED is wrong. It MUST be plugged into PWM6.

moscher, 2017-01-31:
Yes get the LED Signal from PWM6 (J6). 
I can program the LED with all options i see there and it worked fine, but the Indicator does work wrong as mentioned above.

Mikowmer, 2017-02-01:
I think I may be able to shed some light on this issue, and what may be happening.
I have a similar issue. I have an eight LED light bar set up on the back of my Quad. When I set it up like so:
[image]
When I indicate left, only the two left most blink. When I indicate right, the other six blink.
I've also found that picking the orientation of the LED's **_does not work_**. At all.

However, configuring them like this:
[image]
It works fine, 4 left, 4 right. And like this:
[image]
3 left, 5 right. By the looks of things, the indicator is decided by what side of the quad it is on (notice how the grid is split when you enter wire ordering mode). However, I believe that the center of that grid has been improperly assigned to the center, and has instead been assigned... somewhere between 4 and 6 notches from the left of the assignment board.

AndersHoglund, 2017-02-01:
If this can help to shed some light, I have indicators working perfect with an 8 LED strip on SPRACINGF3 BF3.1.1 with this config:

    # version
    # Betaflight / SPRACINGF3 3.1.1 Jan 30 2017 / 16:45:33 (b135d1c)
    
    # led
    led 0 11,15:E:AI:0
    led 1 10,15:E:AI:0
    led 2 9,15::AT:0
    led 3 8,15::AT:0
    led 4 7,15::AT:0
    led 5 6,15::AT:0
    led 6 5,15:W:AI:0
    led 7 4,15:W:AI:0
[image]

moscher, 2017-02-01:
@Eganwp thats a nice discovery.

i tried it out and get my 4 leds to work.
[image]

the direction arrows are not nessesary for me, when putting the leds in the corners.
Is this a bug or wanted?

The grid should be marked what section is for what direction for.

mikeller, 2017-02-02:
Ok, I had a look in the code, and the orientation setup is quite messy indeed (the 'dimensions' of the led arrangement are calculated from the top left corner to the rightmost / bottommost LED).

Also, the 'LED orientation and colour' arrows are only used for the 'Modes and orientation' function, not for the indicator. And they are combined with the 'orientation' on the board there.

I will look into simplifying this, my suggestions on what to do are:
- use real dimensions of the LED arrangement for 'board orientation' (i.e. above middle line of all of the defined LEDs is 'NORTH', and so forth);
- use 'board orientation' only for indicators;
- use only 'arrow orientation' for 'Modes and orientation'.

Your thoughts?

Eganwp, 2017-02-02:
Personally I think column/row # referencing from top left is the best way as it's the industry standard for grid counts. Similar to array usage as well. I wouldn't switch I to NSEW.

mikeller, 2017-02-02:
@Eganwp: I wasn't proposing to move away from top / left referencing. it's more that at the moment the size and midpoint of the LED arrangement is calculated starting from the top left corner (so the size of the 4 LED arrangement in the topmost image comes out as (9, 13). This is what leads to the weird behaviour when moving an arrangement around on the grid. My proposal is to calculate the size / midpoint from leftmost LED used to rightmost, and from topmost to bottommost (i.e. ignoring empty space on the grid).

Eganwp, 2017-02-02:
Aah, I see. Ya that's a good idea and would make it more intuitive for people to use with a simple instruction note on the configurator page.

Mikowmer, 2017-02-02:
I'm assuming that the configurator is set up such that you can just place the led's on the grid where they are on the quad, the idea being that the indicator is then set up around that. Would it be possible to add an over- or underlay the places a frame on the grid, such that you can get a better idea of where the LED's are? And then would having a set centre on the grid image help with placing the centre in the code?

mikeller, 2017-02-02:
@Mikowmer: It would certainly be possible to add the functionality for an underlay, but I don't think this will help much with the inherent complexity of the current LED configuration. Just placing them in the approximate shape of your craft should work just as well. But if you feel like contributing the underlay functionality feel free to do so.

Re setting an arbitrary centre for the grid, that's not possible, since the centre is evaluated at boot time on the craft. And I don't want to add more complexity to the LED configuration.

Mikowmer, 2017-02-02:
@mikeller So... when I turn on my Quad, for whatever reason it believes that somewhere that isn't the centre is the centre point and _that_ is what is causing these issues?

mikeller, 2017-02-02:
Yes. The position of each LED on the grid is stored on the quad as part of the config. On startup, the quad calculates the mid position for the purpose of indicators / orientation colours, and uses the grid origin and the rightmost / bottommost LED that is defined as the edges of the quad, which is plain wrong.

carloscampos775, 2017-02-10:
Still having a issue where pitch back and roll right only activates the indicators.

Mikowmer, 2017-02-11:
@carloscampos775 The commit hasn't been merged back into the master branch yet, so the problem won't be fixed yet.

mikeller, 2017-02-11:
It will be included in release 3.1.6 (#2367), due on 01/03.