betaflight/betaflight #2231: Motor "blip" on board reboot

joshuabardwell, 2017-01-27:
I just upgraded one copter to 3.1 tonight. Getting short motor blips on board reboot--e.g. when saving config or closing the configurator. Thank god props were off (naturally!).

Diff is attached.
[QAV-R.txt](https://github.com/betaflight/betaflight/files/734548/QAV-R.txt)

mikeller, 2017-01-27:
Not sure if spurious signal detection isn't normal during reboots - my rules for craft config are 'either props are off, or battery is off'.

joshuabardwell, 2017-01-27:
Granted. However, there is no reason for this to happen. It has happened in the past, and has been fixed by adjusting some delay or something. Bear in mind that it could happen in the field as well. For example, you could land, then you could be configuring PIDs via bluetooth or WiFi and save config. This is a completely normal and safe thing to do, under normal conditions.

borisbstyle, 2017-01-27:
There is a delay on every boot and shutdown of pwm signal on each reboot commanded from MSP or cli.
That has been proven to work for a long while now and it still does the job. Perhaps this is something in the esc firmware?

MJ666, 2017-01-29:
For the BlueJayF4 and AlienFlightF4 brushless the ESC's reinitialize every time if you save configuration even if the FC did not reboot (ACC calibration, saving PID's). This looks to be specific for F4 targets and I can not reproduce with an F1 or F3 based FC. Motor are not starting. This also happens if you calibrate with stick commands. This is also there for BF 3.0.0. Usually I do all configuration via USB and don't have a battery connected. Also don't use Bluetooth or an OSD to make changes in the field. Using an OTG cable with EZGUI and my Andriod phone.

SurgeonFPV, 2017-01-29:
I opened an issue about this as well. Last night after making changes while battery plugged in, FC rebooted. One motor blips and smoke came out of it. Never had that issue before. FC is a Revo f4