betaflight/betaflight #2233: Motors tab defaults to "Master Slider Enabled" and sliders active when DSHOT600 is selected

joshuabardwell, 2017-01-27:
I upgraded from 3.0.1 to 3.1 tonight. I initially configured with multishot. Then I switched to DSHOT. When DSHOT is selected, the Motors tab defaults to having the "master slider" enabled and the sliders are active. Switching back to multishot reverts to normal behavior.

Configurator version 1.9.0, betaflight 3.0 release version. Target is Fury F3 if it matters.

borisbstyle, 2017-01-27:
This should go to configurator issue list https://github.com/betaflight/betaflight-configurator

But it was somehow tricky to get dshot working on the sliders from what I remember as it doesnt use 1000-2000 range at all.

@mikeller Can you check this?

joshuabardwell, 2017-01-27:
Sorry, you're absolutely right. I forgot there is a separate repo for the configurator.

mikeller, 2017-01-28:
@borisbstyle: Chur!

mikeller, 2017-01-29:
Erm, @joshuabardwell, did you mean Betaflight 3.1? 3.0 doesn't have DShot.

Even with 3.1, I wasn't able to reproduce. I selected DShot 600, then 'Save & Reboot', after reconnecting I went to the Motors tab, but the sliders were at 1000 and the 'Safety Switch' off.