betaflight/betaflight #2234: Add option for enabling "always-on" blackbox logging.

ckwkalk, 2017-01-27:
At the moment there is the option to enable blackbox logging on motor test [blackbox_on_motor_test] but this requires either the GUI (via USB) or an external trigger from the controller to be activated. If there was an option to automatically enable "blackbox logging on power up" it would make it much easier to check the sensors (GPS gyro etc..) with a minimal setup . My own interest is in comparing the performance of different boards in a repeatable way (see image of calibration rig below) but I'm sure there are many other use cases where better control of the blackbox logging would be useful? 

[image]

http://forum.arduino.cc/index.php?topic=145781.0

borisbstyle, 2017-01-29:
Its much easier for you to modify the code for that as this is not a "standard" usage :D

borisbstyle, 2017-01-29:
but you can easily use Bluetooth connection to enable / disable it

ckwkalk, 2017-01-30:
@borisbstyle The bluetooth + GUI solution gives me a start thanks. I'd still like to "trick" the board to automatically go into blackbox logging mode as soon as the battery is connected. I'm happy to modify the code but what is the best way to fool the system into thinking it is powered up (safety is not a concern as it will not be connected to motors)?

mikeller, 2017-01-30:
@ckwkalk: If you give me a couple of days I can look into extending `blackbox_on_motor_test` to allow 'normal' / 'motor_test' / 'always' logging.

ckwkalk, 2017-01-31:
@mikeller that sounds great; and hopefully will be of interest to others as well.

ckwkalk, 2018-02-09:
@mikeller just wondering if anything has changed in betaflight that would allow me to continuously log sensor output as soon as the system is powered up? You mentioned it might be possible to extend [blackbox_on_motor_test] to allow 'normal' / 'motor_test' / 'always' logging. Thanks!

mikeller, 2018-02-10:
Yes, sorry, that has been implemented in #4498. `set blackbox_mode = always`.

ckwkalk, 2018-02-14:
Thanks for that - I am grateful - will have to update and check it out.

closed 2018-02-14 by ckwkalk