betaflight/betaflight #2238: 3.1 Piko version on Radiance and Kombini, Transponder has stopped Working

FPVDjp, 2017-01-27:
Release 3.0.1 Transponder flashes and works correctly on these boards, updated to 3.1 and the LED no longer illuminates at all.  Also when enabling Transponder in configuration Tab the boards will reboot in to DFU mode.

jflyper, 2017-01-27:
Are you using DSHOT?

FPVDjp, 2017-01-27:
No using Multishot

jflyper, 2017-01-27:
Oh, okay, then there's something else...

FPVDjp, 2017-02-01:
Flashed all my Piko boards to 3.1.1, issue still exists, reverted to 3.01 and they work again, then I tried the release candidates, up to 3.1 RC8 and it worked, seems to stop working from 3.1 RC9 onwards.

mikeller, 2017-02-02:
@FPVDjp: Not sure if the old style resource definition for the transponder is still supported in 3.1. I've converted the definition to the new format, but I don't have a board so I can't test. If you could test the attached binary and report back if this works? (Also, after configuring your board, please reboot, then go into the CLI and do a `resource list` and confirm that it contains `B08 TRANSPONDER`.
[betaflight_3.2.0_PIKOBLX.zip](https://github.com/betaflight/betaflight/files/746588/betaflight_3.2.0_PIKOBLX.zip)

FPVDjp, 2017-02-02:
@mikeller yes the resource list does show B08 Transponder, but still no illumination from the LED with the 3.2 hex supplied in the zip file unfortunately.

mikeller, 2017-02-02:
Ok. The problem must be somewhere else then. Without a board to test I cannot help you much more, sorry.

Gripchion, 2017-02-19:
@FPVDjp Did u get the transponder to actually work with an I Lapse system on Betaflight 3.0.1? I never could get my I lapse to pick up the code. I had the same problem of the ir led not flashing when I upgraded to latest version also.

FPVDjp, 2017-02-20:
@Gripchion yes it works for me with 3.0.1, are you using the CRC code in the Betaflight GUI as it does not work using a standard 7 digit numerical value?

Gripchion, 2017-02-20:
Yes I bought a code from sp racing. Code example would be hex decimal, 65aebb2a3c10. My ir led is flashing, confirmed with cell phone camera. I scan for new pilot with I lapse system. It will not pick it up. The I Lapse system will pickup a stand alone transponder that is bought from I lapse. Is there any other settings in betaflight that need to be turned on beside the transponder tab. Maybe a uart/ LED. I messed with this for about 3 weeks of and on. I'm at a loss. You are the only one that I have ever heard of getting it to work. I have 3 different Kombini boards. Couldn't get any of them to work. Maybe a setting in I lapse????????? If u can help me to get this to work, I would buy u a steak dinner and a beer or 2.😆

  Transponder has stopped working (#2238)

FPVDjp, 2017-02-21:
I had to try a number of different codes before I found one that worked reliably, try some others, you can generate some here:  http://owenson.me/ilap/

JBFUK, 2017-05-10:
Still not fixed?

bumpyconstant, 2017-05-10:
Doesn't seem to be working yet. Hopefully this will be fixed soon. I need transponder and smartaudio together.

mikeller, 2017-05-10:
As stated above, none of the developers in here seems to have access to one of these boards, making it impossible to test / fix this. Maybe you should contact the supplier of the board and ask them to look into fixing this?

JBFUK, 2017-05-12:
I requested that FuriousFPV provide one of the devs with a board.  Hopefully they will be able to help with that.

mikeller, 2017-05-13:
Thanks, @JBFUK.