betaflight/betaflight #2241: make arm_sdk_install broken?

fishpepper, 2017-01-27:
It seems like the arm sdk download location was moved to arm:
https://developer.arm.com/-/media/Files/downloads/gnu-rm/5_4-2016q3/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2

changing the url base to 
ARM_SDK_URL_BASE  := https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926
seems to fix this

The old launchpad site seems to be down.

mikeller, 2017-01-28:
The source URL was moved to developer.arm.com for 6.2-2016q4, since the launchpad project states that they have stopped adding new releases.

AndersHoglund, 2017-01-28:
Yes, but this is about the old 5.4. And I can't see the problem. 5.4 are still installed from launchpad on the patch_3.1.1 branch while 6.2 is installed from arm.com on master branch. Just tested both, no problems.
@fishpepper please re-test and close the issue if it works for you. You might have been hit by a temporary outage.

fishpepper, 2017-01-28:
ok, now it works again. yesterday i got a file not found error for 5.4

closed 2017-01-28 by fishpepper