betaflight/betaflight #2243: SP Racing F3 Mini Magnetometer not detected in v3.1.0

mrbob1234, 2017-01-28:
Hi all. I'm still relatively newish to all of this but thought I would just make you aware of a problem I came across. After flashing betaflight for the first time on one of my micro quads and quite liking it, I thought I would give it a go on the SpF3 Mini (v2) I just got. I flashed the 3.1.0 for the SpF3 mini with no problems and everything seems to be working apart from the fact I can't get the magnetometer to work! I have ofcourse selected it and the barometer (which works without any problems), but it just doesn't become active on the top sensor bar. I have tried reflashing it a few times without success, but have noticed that it does work and becomes active without any problems on 3.0.1 and also ofcourse in cleanflight. Just thought I would make you aware, nothing much came up on my searches etc. 

Thanks alot for all the hard work you guys do, so people like me can get into this kind of hobby!!

jflyper, 2017-01-28:
I didn't know they had (v2) variant of the SPRACINGF3MINI. Any pointer?

mrbob1234, 2017-01-28:
Apparently the main differences in the v2 is a new BEC that isn't so predisposed to blow and removing the balls...!!! 

Although, I had a v1 a while back and the gyro was broken when I got it, which wasn't the case with this one, so that is nice!!! 

Ahhh, that would make sense then! It was the 3.1.0 version indeed, and it worked on v3.0.1, so that's that! Where is that listed and what were the reasons behind deleting it? And will it be back anytime soon? Not sure whether to go back to clean flight or just stick with it now... I am getting comfortable in rate/air mode, so probs won't even need it to be honest, I presume it is only used for heading lock?

jflyper, 2017-01-28:
Nono. The support for MAG has been deleted UNTIL end of the december. So, if your 3.1.0 (pre) is older than a month, you may have a binary that doesn't have the mag support.

mrbob1234, 2017-01-28:
I literally just got the board yesterday! It ofcourse arrived with cleanflight and I flashed it with BF 3.1.0 using the betaflight configurater. Made sure to select full chip erase and even re-flashed it several times. I then tried 3.0.1 and the mag worked in that, but again would not come on when I re flashed the 3.1.0. Reverting back to cleanflight today it is working in cleanflight...

jflyper, 2017-01-28:
Okay, I believe there is a problem in mag detection code for 3.1.0.

mikeller, 2017-01-28:
Hi @mrbob1234. Trying to debug this here, but I don't have an SPRACINGF3MINI board to test, so I'll need some assistance. First, can you please flash 3.0.1 onto your board, enable all the sensors in the 'Configuration' tab, and then send me the output of `status` in CLI? This will give us the information what magnetometer hardware is actually on your board.

mrbob1234, 2017-01-29:
Hi @mikeller , no problem, here it is: 

System Uptime: 12 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:7%
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8975
Cycle Time: 507, I2C Errors: 1, config size: 1520
SD card: Manufacturer 0x3, 15159296kB, 10/2016, v3.0, 'SL16G'
Filesystem: Ready

Let me know if there is anything else I can do!

mikeller, 2017-01-29:
I couldn't find anything obviously wrong with the SPRACINGF3MINI target in 3.1, but some things could be cleaned up. Can you please try and see if Magnetometer is detected with this firmware:

[betaflight_3.2.0_SPRACINGF3MINI.zip](https://github.com/betaflight/betaflight/files/737443/betaflight_3.2.0_SPRACINGF3MINI.zip)

mrbob1234, 2017-01-29:
@mikeller unfortunately no luck! 

This is the status now: 
# status
System Uptime: 22 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
SD card: Manufacturer 0x3, 15159296kB, 10/2016, v3.0, 'SL16G'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 1624
CPU:7%, cycle time: 508, GYRO rate: 1968, RX rate: 49, System rate: 9

mrbob1234, 2017-01-29:
`# status
System Uptime: 22 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
SD card: Manufacturer 0x3, 15159296kB, 10/2016, v3.0, 'SL16G'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 1624
CPU:7%, cycle time: 508, GYRO rate: 1968, RX rate: 49, System rate: 9`

RAMukOK, 2017-01-30:
FYI... I'm having same issue.  SPR F3 Mini.  Just flashed 3.1.0 with 'load firmware (online)', tonight, and no mag.

in case it helps...

# version
# Betaflight / SPRACINGF3MINI 3.1.0 Jan 25 2017 / 10:00:11 (88693ca)
# status
System Uptime: 408 seconds
Voltage: 1 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
SD card: Manufacturer 0x22, 3917824kB, 03/2016, v0.1, 'MSD20'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 1624
CPU:24%, cycle time: 259, GYRO rate: 3861, RX rate: 49, System rate: 9

RAMukOK, 2017-02-04:
Anything new on this issue?

gramk, 2017-02-04:
I am having the same issue, magnenometer icon is off since 3.1.+

joshendy, 2017-02-04:
I'm also having this issue.

Status from cleanflight v1.14.2 - SPRACINGF3 - 2016-12-31

# status
System Uptime: 12 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.18
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
PIDd: 449, GYROd: 448, I2C Errors: 0, registry size: 52

Status from Betaflight / SPRACINGF3 3.1.3 Feb  2 2017 / 13:58:33 (73b498e)

# status
System Uptime: 10 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 1604
CPU:7%, cycle time: 500, GYRO rate: 2000, RX rate: 49, System rate: 9

RAMukOK, 2017-02-04:
FYI, just tried the latest available firmware (in the configurator, 'show unstable'checked...
Still no MAG.

# version
# Betaflight / SPRACINGF3MINI 3.1.3 Feb  2 2017 / 13:59:45 (73b498e)

# status
System Uptime: 136 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
SD card: Manufacturer 0x22, 3917824kB, 03/2016, v0.1, 'MSD20'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 1612
CPU:27%, cycle time: 251, GYRO rate: 3984, RX rate: 49, System rate: 9

emma0223, 2017-02-06:
betaflight/spracingf3mini 3.1.3
status
System Uptime: 68 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
SD card: None inserted
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 2, config size: 1612
CPU:9%, cycle time: 501, GYRO rate: 1996, RX rate: 49, System rate: 9

emma0223, 2017-02-08:
Still having te same problem with version3.1.5

borisbstyle, 2017-02-08:
When release notes dont say anything about it it means its not fixed unfortunately for you.

The mag, baro and gps are really best effort support in betaflight. Those are also not being tested. Usually it was working before ae betarlight was really close to cleanflight, but since the code changed a lot compared to previous releases it became broken.
It is really down in the priority list for now unless someone is there who is able to put some time and effort in it.

If you really badly need the mag mode than I suggest trying iNav instead of betaflight.