betaflight/betaflight #2244: MSP over s.port - PID values won't survive power cycle

nitbeat, 2017-01-28:
Hello,

I flashed the new 3.1 firmware on target: SPRACINGF3
Telemetry works, PID Values are shown and I verified that when I change value it also changes in the
flight controller (via configurator PID Tab)

But when I power off power on, the values return to the previous numbers :(

I'm using this LUA Script which was recommended by one of the dev people 
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/BFSetup.lua

I heard from somebody that he used another LUA script and it fixed the problem.
any ideas ?

raphaelcoeffic, 2017-01-28:
@nitbeat you should really check that the correct script is executed. I've seen a case where an old script version was ghosting on the SD card and was still executed on the telemetry screen. Start trying to execute the last version directly through the SD card explorer. This way, you can make sure which file is exactly executed.

nitbeat, 2017-01-28:
Alright I'll all the script and use this one:
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/BFSetup.lua

Is this the latest one ?

raphaelcoeffic, 2017-01-28:
Yes, is is.

raphaelcoeffic, 2017-01-29:
@nitbeat Just a side remark, as I noticed that you have port sharing between SPORT and MSP.

Please note that connecting SmartPort to your MSP port might corrupt your MSP traffic when connected through USB.
SmartPort receivers poll for all sensor IDs continuously, so that it may interfere with the MSP traffic.

And another one concerning this specific issue: once armed, you cannot save parameters into EEPROM. That would not work either if you did it with the configurator.

si618, 2017-01-31:
@raphaelcoeffic, I had the same problem when hooking SmartPort up to TX on UART 1 (shares USB on my X-Racer 3.1 FC). It would only show PIDs and rates when armed, and wouldn't save them, which makes sense now :)

PIDs and rate changes are working fine on UART 2, still trying to get VTX working for my tramp, but just saw your other comment about VTX support missing from 3.1.0 stable, so will try a nightly build.

With UART 1 apparently out of action, and UART 3 using sbus, I'm relying on soft serial for tramp telemetry, so hopefully that will work!

nitbeat, 2017-01-31:
UPDATE - "P.Y." just posted in my guide:
http://www.nitbeatfpv.com/frsky-taranis-pid-settings

A brilliant solution IMO for fixing the saving problem.
If you have to use UART 1, after you save the values using LUA script,
use your STICKS TO INPUT A SAVE COMMAND
LeftStick - down + left
RightStick - down + right 

I verified it works.

Although, Using UART 2 is much better as the values appear without having the need to ARM and YOU DON'T HAVE THE SAVING PROBLEM :)

Thanks all for your help, I will add this info to my guide 

CHEERS !

raphaelcoeffic, 2017-01-31:
Cool! I think that helps a lot of people!

-Raphael.

PS: hope you'll get some working Tramp soon. They seem to be sold out everywhere, as IRC is preparing the second batch.

nitbeat, 2017-01-31:
Yeah ! that's really cool :)

BTW, is there any reason this method won't be supported in the new FRSTY RC ?
FrSky ACCST Taranis Q X7

I know it will support it but just making sure :)

raphaelcoeffic, 2017-01-31:
I'd love to support the smaller screen on the QX7 as well, but don't have any to test/try. I'd love to get one, but they're hard to procure at the moment. Maybe if somebody owning one would help me... I'd need a working simulator to sketch the screen layout though....

nitbeat, 2017-01-31:
so the feature will work, you just need to adjust the layout in the Script
code for the QX7 ?

raphaelcoeffic, 2017-01-31:
yes, I think so (as far as I know).

merinkumar, 2017-03-23:
Does the lua script work with D4R II? am able to get all sensors reading in the telm screen (12) on my QX7 but when i go to PIDS screen, only layout is there no PIDS, bottom it says NO TELEMETRY.

SoulPictures, 2017-03-23:
The LUA communication is SMARTPORT based so it should only work with X-Series receivers and not with the D4R

merinkumar, 2017-03-24:
i had the same question, i started my work with this video, its an tar old but uses d4r ii an x8d. So may be earlier the support was there and now stopped ?

mikeller, 2017-03-24:
No. D4R only supports the FrSky hub protocol, and not SmartPort. Since FrSky hub only transmits data in the direction flight controller -> TX, there is no way that it can be used to send settings from the TX to the flight controller.

merinkumar, 2017-03-24:
thanks, that makes sense...