betaflight/betaflight #2246: Funktion Telemetrie Hott does not work

Nexus1212, 2017-01-28:
Hello,
the connection between Flightcontroller and hott receiver does not work.

Release: 3.1.0 OMNIBUSf4SD (stable)
Target: FLIP 32 - F4 - OMNIBUS V2 PRO
Receiver: GR-12sh+ 

Receiver testet on Racing SP3 with Cleanflight. Telemetry works fine! 

I connect the UART6 RX and UART6 TX to Channel 6 (telemetry).
Between UART6 TX and Channel 6 is a 1N4148 diodes with cathode marker to UART6 TX.
(https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md)

The problem is in my RC-Transmitter is no option "ON" for EAM or GAM.

I think the connection does not work between OmnibusF4 and GR-12sh+. Du you need more information?

One more Question: Do you think about a alert for low battery capacity as the alert for low battery voltage?

Best regards
Nexus

jflyper, 2017-01-28:
I think F4 targets do not support HoTT telemetry without an external inverter.

Nexus1212, 2017-01-28:
Hello,
I have no experience with an external inverter.  Is it the same problem as the Frsky inverter?

I think the F4 have 3 "normal" UART`s. Why is a external inverter needed?
Could you send me link of a required component?

blckmn, 2017-01-29:
No, but UART6 possibly has a buffered (inverted) input for use with frsky etc.

closed 2017-01-29 by blckmn