betaflight/betaflight #2247: VTX Channel from OSD not displayed when smartaudio TBS

michelmorin, 2017-01-28:
I'm using smartautdio TBS VTX which I can change successfully in betaflight 3.1 (BrainFPV fork), but when I choose VTX channel in the OSD config in betaflight, I was expecting that the VTX channel and frequency would get displayed on the OSD, but it is blank.

From RCGroup, Martin from BrainFPV checked the code and said it's currently setup to show from boards that  have built in VTX on them.    Here the link
https://www.rcgroups.com/forums/showpost.php?p=36748840&postcount=473

Thanks,
Michel

raphaelcoeffic, 2017-01-28:
That's right. We still need to integrate all the mechanisms various VTXs use to be available for all of them.

jflyper, 2017-01-29:
#2254 should display band and channel in CH:bc format (e.g., CH:R4, CH:F1).

For frequency display, I don't think it should be a separate display item, since band and channel would suffice the purpose of this display item most of the time.

However, the item may displayed in frequency if the device is not operating in band-channel mode (e.g., SmartAudio and Tramp has frequency mode; but this mode is not handled yet).

michelmorin, 2017-01-29:
I agree, showing CH:bc format without frequency would satisfy my needs of having this displayed in the OSD

bnn1044, 2017-01-30:
I think we should give option for display either frequency or CH and band

jflyper, 2017-01-30:
@bnn1044 Frequency can be added very easily and would show off, but majority of the users don't need them.

bnn1044, 2017-01-30:
@jflyper. Yes. I know it is easy. And some of them need to know the frequency. Like for organized fun fly with a. Few friends, some people like to know the frequency. Some like to know the band and ch.  that is why I say make it a option.