betaflight/betaflight #2251: Using resource mapping to fix Cc3d motor layout

dronejunkie, 2017-01-29:
just tried the remap on my cc3d motor layout, it flipped to the left during take off.  Here was the mapping before the change with custom mix.
resource MOTOR 1 B09
resource MOTOR 2 B08
resource MOTOR 3 B07
resource MOTOR 4 A08
resource MOTOR 5 B04
resource MOTOR 6 A02

I have changed it back to quad x default mix and below is the new resource mapping. I have saved the change and here are the values after a reboot
resource BEEPER A15
resource MOTOR 1 A08
resource MOTOR 2 B08
resource MOTOR 3 B09
resource MOTOR 4 B07
resource MOTOR 5 B04
resource MOTOR 6 A02

Please clarify if it is a bug.

dronejunkie, 2017-01-29:
I am using 3.1.0 final release

mikeller, 2017-01-29:
I'd try standard wiring, with mixer QuadX, and default resource mapping.

dronejunkie, 2017-01-29:
I just wanted to make use of this new feature. Since then I rolled back to
the custom mixer. I raised this issue to check if it was something I did is
incorrect or it is a bug. I do see the value of your suggestion, which I am
assuming is to make sure it works properly first and then try again.
Unfortunately I got the original falcon 250 cc3d which requires desoldering
all of pin headers on the cc3d to rewire. A lot of work.

Anyone could clarify what I did is correct is sufficient for now.

mikeller, 2017-01-29:
Even then, without knowing what motor layout / custom mixer your quad has, telling if your remapping is correct or not is impossible.

dronejunkie, 2017-01-29:
Here is the motor layout
1 2
4 3

And the custom mixer which works with the above layout is

*mmix 0 1.0, 1.0, **-1.0**, **-1.0*
*mmix 1 1.0, **-1.0**, **-1.0**, 1.0*
*mmix 2 1.0, **-1.0**, 1.0**, **-1.0*
*mmix 3 1.0, 1.0, 1.0, 1.0*

mikeller, 2017-01-29:
Try this:

```
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 1 B07
resource MOTOR 2 B08
resource MOTOR 3 A08
resource MOTOR 4 B09
```

dronejunkie, 2017-01-29:
I c, swap 1 to 3,  4 to 3, etc ..  Will try tomorrow. Thanks

AndersHoglund, 2017-01-29:
There is a correct "Quad X 1234" now in 3.1, for Revolt and OP/LP motor mounting order. Reset to defaults and select that mix. No re-mapping needed.

dronejunkie, 2017-01-29:
Verified it is working.  mikeller thanks mate. 
@AnderHoglund, I have not verified your suggestion but it is certainly less work. 
No issue just someone who does not know what he was doing.

closed 2017-01-29 by dronejunkie

mikeller, 2017-01-29:
@AndersHoglund: n.b. Betaflight always had a QuadX1234 mixer, and it always was correct.

AndersHoglund, 2017-01-29:
The mixer might have been correct in BF, but the BFC image was not. Usabiltity you now. End user communication. Ref: https://github.com/betaflight/betaflight-configurator/pull/399