betaflight/betaflight #2252: BF 3.1 - Failsafe not working with airmode on, motors spin at airmode speed

SurgeonFPV, 2017-01-29:
Hi, i just flashed BF 3.1 on my Revo F4. I've had no issues with any other BF version and set it up the same as always.
I have arm and air mode on separate switches, however this applies even if i enable airmode permanently or on the same switch as arm.
If i arm the quad, motors spin, when i switch off the radio, i can hear the motors change speed slightly but they keep spinning at airmode speed or so. I have waited at least 5 seconds. Even though failsafe is set to 1 second.
If i do this without airmode on, i just give it a little throttle and then switch off the radio. it works no problem. Please help, i may flash back

howeson1, 2017-02-01:
Tested on impulsef3 with airmode permanently on and unable to reproduce

howeson1, 2017-02-01:
If your flight controller has blackbox then im sure that would help people sort it out for you

closed 2017-02-01 by SurgeonFPV

howeson1, 2017-02-01:
Was it user error or can you not reproduce?

SurgeonFPV, 2017-02-01:
Something changed at some point. I got around it by setting failsafe on my airmode switch.