betaflight/betaflight #2253: BF 3.1 - Motor smoke after enabling Airmode permanently

SurgeonFPV, 2017-01-29:
Hi, i just flashed to BF3.1 and was setting things up. I had the battery in and i went into the configuration to enabled airmode permanently. When i hit save and reboot. One of the motors twitched and smoke came out. Motor seems fine but is this a bug?

AndersHoglund, 2017-01-29:
Never ever configure or reset or restart the FC with ESCs powered up. 
There are other cases like this, e.g. ref: #2231

waltr1, 2017-01-31:
Correct. OR:
**ALWAYS use a Current Limiter whenever connecting a LiPo on the Bench for Testing**, ESC Cal or Flashing or any time the Config GUI is opened.
https://www.rcgroups.com/forums/showthread.php?2327875-DIY-SAVE-YOUR-ELECTRONICS!-BUILD-A-SmokeStopper%C2%99-!

closed 2017-02-02 by borisbstyle

cschneider, 2020-01-18:
Very good advice. Could you make sure to document that at a prominent place in the betaflight UI? I think powering the ESC without a current limiter while configuring also burnt two of my motors.