betaflight/betaflight #2255: AirbotF4 switch inputs processed despite not being active

RipperDrone, 2017-01-29:
AirBot F4 on BF 3.1 Official Release detects switch 'AUX 3' inputs from my Taranis X9D+ despite having set this specific switch only active in case 'AUX 4' has to be in the range above 1025 (see picture below). Seems like the 'do nothing' command is not recognized properly.

On another PikoBLX copter with the same adrange settings everything is normal. Can this be a target specific  bug? I even compared the CLI dumps, all adrange entries are exactly the same. PikoBLX does not react to AUX 4 inputs while AUX 3 is in range below 1025, whereas AirbotF4 does...

[image]

AndersHoglund, 2017-01-29:
Sure it can be different by target, or rather by Rx . With some Rx you only only have 6ch, some 8,10,12 etc. You might not have AUX 3 and 4 on all rigs, check in the Receiver tab. 
There is also the `max_aux_channels` setting, might be set differently.

RipperDrone, 2017-01-29:
@AndersHoglund Yes, this was one of the things I checked:
- max_aux_channels = 5, for both PikoBLX (works ok) and AirbotF4 (buggy)
- Same RX onboard: FrSky XSR

In BFC, all RC channels / switches flipped are showing ok in the Live Remote Preview tab. When switching AUX 4, the yellow position indicator in the switches tab moves ok as well.

Must be something else...

mikeller, 2017-01-29:
Could this be a case of looking in the wrong place, i.e. is there an adjustment function defined *at the bottom of the table, outside of the visible area in the screenshot* that covers the range below 1025 as well?

RipperDrone, 2017-01-29:
@mikeller Guess this would have been too easy:

[image]

RipperDrone, 2017-01-29:
This is what the RC screen looks like (quad powered and Taranis switched ON). When I change AUX 3 from 1500, FC beeps / processes a valid AUX 3 command despite AUX 4 still staying @ <1025, so AUX 4 should be reliably 'locked'.

[image]

AndersHoglund, 2017-01-29:
What is the real world symptoms? In what way is this issue a problem? Accidental AUX3 activation? But as the action is set to "No changes" nothing bad can happen. Or?

RipperDrone, 2017-01-29:
@AndersHoglund AUX3 is assigned to a potentiometer knob (all digital 3pos switches are assigned to other commands) so it often activates accidentally, e.g. when simply taking the Taranis RC out of my backpack. This triggers a beep accompanied by blue LED flash on the AirbotF4 FC, and probably changes some PID parameters unintentionally at the same time.

But now that you brought up the question about what can happen while "No Changes" is set: I'm not sure which PID settings - if any - are actually changed, but blue LED flash / permanent buzzer beeps seem to indicate something is being changed. 

As mentioned above: Works ok on all F3 FCs like SPRacingF3 or PikoBlx. Therefore, it seems to be an Airbot F4 specific bug...

All my model settings on Taranis are the same except receiver numbers. So are all CLI settings regarding switches / adjrange. Really clueless here...

One thought about F3 vs. F4 differences: AirbotF4 issues buzzer/LED signals while connected to USB only - which the F3 SPRacing or PikoBLX don't. Maybe the F3s do react on the AUX4/AUX3 switch setting in "No Changes" mode as well, but there is no LED/buzzer signal triggered on F3s? However, when switching AUX4 to >1025 and then adjusting PID parameters by rotating the AUX3 knob up or down from 1500, both F3s and F4s sound slow beeps while lowering values, faster beeps while increasing them.

AndersHoglund, 2017-01-29:
OK, I saw the difference in F3 vs F4 behaviour. But I needed to ask anyhow.... 
I also have adjustment settings. Two 3-way switches for PID and RPY selection and another 3-way for pumping values up/down. I do not have any "No Change" action. Happens sometimes that the "value pumper" is not in centre position. Then I just set it there, and re-start the rig. No biggie.

RipperDrone, 2017-01-29:
Yep, it's no biggie to live with it. Just noticed the difference btw. F3s and F4s, respectively. Was expecting the "No Change" setting to actually mean and be "No Change" and to work the same way on F3s and F4s :-)

AndersHoglund, 2017-01-29:
Wait. Is the problem that you do NOT get any beeper notifications on some FCs? I might have misunderstood the problem. Easy fix, remove that "No change" action.  
I certainly do want to be notified when my switch is not centred.

RipperDrone, 2017-01-29:
No, sorry for not making it perfectly clear.

- on F3s, I am getting **no** LED/buzzer signals as long as AUX4 is below 1025 (supposed to act as a "no change unless this 3pos switch is flipped to active). When switching AUX4 to higher values, I am getting signals when moving AUX3 out of neutral. 
- on F4s, I am getting LED/buzzer signals whenever AUX3 is moved out of neutral, regardless of AUX4 setting. 

F3s' behavior is what I want, F4s' is not. 

Since AUX4 is a potentiometer switch, it's quite easy to accidentally move it out of neutral position when handling the RC...

Side note: AUX4 is a combo switch created on the Taranis to provide the fine steppings found in the adjrange commands I highlighted earlier in this post.

AndersHoglund, 2017-01-29:
Why would you NOT want to be notified when the AUX3 pot is not in neutral? I would want that.....

RipperDrone, 2017-01-29:
Ok, maybe that's a way to look at it. 

Anyway, in the first place I'd like to be notified when values are being changed. When buzzer sounds/LED signals are always triggered when AUX3 pos is not in neutral (irrespective of AUX4 being in No Changes position or in any PID parameter position), it's hard to tell by the LED and buzzer sounds whether parameters are being changed or not.

AndersHoglund, 2017-01-29:
Remove "No Change" action, then all beeps and blinks will mean a change. 
I am more and more thinking this is a non-problem.... Well OK, there might be an issue with "No Change" acting different on different targets.

RipperDrone, 2017-01-30:
Yes, that's the point. Inconsistent behavior throughout different targets leaves a certain feeling of confusion and will trigger noobs' questions like mine :-)

Same with Buzzer sounding 10x init beeps on USB-connect (no LiPo connected) on F4 Airbot, but doesn't on F3 targets (this is because of the hardware boot / reset order and buzzer init seems to power up differently).

So I thought it's best to try make those features as uniform as possible throughout targets...