betaflight/betaflight #2257: BFF3 incorrect default values in BF3.1 and BF3.1.1

AndersHoglund, 2017-01-29:
When flashing BFF3 with 3.1 release version, or current HEAD on 3.1.1, with "Full Chip Erase" or by clicking "Reset Settings" there are some strange default values on some parameters. Diff command prints them despite the reset to defaults. Same thing if using the cli "defaults" command.  
Some values are even out of range, giving error messages when restoring configuration from a saved diff.

    # diff
    # Betaflight / BETAFLIGHTF3 3.1.1 Jan 29 2017 / 14:39:16 (59a4739)
    
    map AETR1234
    set osd_rssi_pos = 1046
    set osd_gps_sats_pos = 1410
    profile 0

    rateprofile 0
    rateprofile 0

Restoring config gives this:

    # set osd_rssi_pos = 1046
    Invalid value
    Allowed range: 0 - 1023
    
    # set osd_gps_sats_pos = 1410
    Invalid value
    Allowed range: 0 - 1023

borisbstyle, 2017-01-29:
Yeah thats actually something IMO incorrectly implemented in the OSD code. The EEPROM gets altered when there is no GPS fix and no RSSI. The values you see are position and status FLAGS combined

borisbstyle, 2017-01-29:
@jflyper Whats your opinion on this? I think OSD code should nog be updating EEPROM values

jflyper, 2017-01-30:
@borisbstyle Let me take a look into it.

jflyper, 2017-01-30:
Ah, it's sorta "Implementation feature".

Position and attributes of OSD display items are encoded into a single 16-bit value:
https://github.com/betaflight/betaflight/blob/master/src/main/io/osd.c#L71-L75
```
// Character coordinate and attributes

#define OSD_POS(x,y)  (x | (y << 5))
#define OSD_X(x)      (x & 0x001F)
#define OSD_Y(x)      ((x >> 5) & 0x001F)
```
https://github.com/betaflight/betaflight/blob/master/src/main/io/osd.h#L22-L23
```
#define VISIBLE_FLAG  0x0800
#define BLINK_FLAG    0x0400
```
Then first call to `osdUpdateAlarms` will add `BLINK_FLAG` to two items, rssi and gps sat count, giving items extra 0x400. This is why it shows up in the diff.

There is no simple way to fix this, except for allocating an array of attribute bits for display items separate from positions so they don't show up in cli. I also think (min,max) for `osd_pos_*` is somewhat pointless, because values are encoded.

martinbudden, 2017-01-30:
@jflyper , could you not call `osdResetConfig` on disarming? Would that not reset them so that they don't show up in the diff?

jflyper, 2017-01-30:
@martinbudden It's not the `osdResetConfig` that is adding a `BLINK_FLAG`, but `osdUpdateAlarms`. You don't want to suppress RSSI and GPS SAT COUNT alarms while disarmed, right?

martinbudden, 2017-01-30:
But `osdResetAlarms` just seems to turn off the blink flags.

jflyper, 2017-01-30:
I think it's best to create a bit encoded array of dynamic attributes for each display items.

It's PITA that we can't use the excess bits in the encoded x,y coordinate variable, but since there are only 21 items (elements) now, so two 32-bit words will do the job. We can make it an array for future expansion also.

jflyper, 2017-01-30:
```
static uint32_t blinkState[(OSD_ITEM_COUNT + 31)/32];
#define SET_BLINK(item) { blinkState[(item)/32] |= (1 << ((item) % 32); }
#define CLR_BLINK(item) { blinkState[(item)/32] &= ~(1 << ((item) %32); }
#define IS_BLINK(item) (blinkState[(item)/32] & (1 << ((item) % 32))
```

jflyper, 2017-01-30:
So the solution for this issue is to

(1) Apply the bit array for blink state. Blink state is NOT a part of the configuration and should have not been encoded as a part of the configuration variables. This change will prevent autonomous update of relevant variables that magically appears with diff.

(2) Visible flag is left as is. Max-min values do not have any meanings.

(2a) We can state this in Cli.md (which only a handful of people take a look at), or
(2b) Create a new flag MODE_ENCODED to print "Encoded value" instead of min-max range.

martinbudden, 2017-01-30:
1) Agreed. Blink state should never have been part of the config variable.

I'd leave the flag as it is. It's prime usage is not for people to be able to set it manually. It's more for backup/restore. Ie do a CLI diff, save the values, upgrade and then apply the diff. The fact that the flags values is not really human readable is not really that important. Certainly not worth the cost of extra code in the CLI.

jflyper, 2017-01-30:
@martinbudden Ack. I'll do the BLINK state changes.

mikeller, 2017-01-30:
@jflyper: All agreed to the above, but can you increase the min / max values so that states with the visible flag on are within the limits? Otherwise restore for values with visible flag on will fail.

AndersHoglund, 2017-01-30:
OK, OSD values fixed.

    # diff
    # Betaflight / BETAFLIGHTF3 3.1.1 Jan 30 2017 / 15:03:26 (35b1941)

    map AETR1234
    profile 0
    
    rateprofile 0
    rateprofile 0

How about `map` , `profile` and (dual) `rateprofile` ? 
After a "Reset Settings" I would expect the diff output to be empty. Or?

 
#

mikeller, 2017-01-30:
The `map` part is a bug.

`profile` and `rateprofile` are in there to indicate the current profile / rateprofile numbers that are being diffed, in order to be able to re-run the output for a restore. Since the dump / diff work straight to output there is no way to tell if they are needed (because there are changed settings) or not at the time they are printed.

closed 2017-01-31 by borisbstyle